KiCad/kicad-footprints #1526: Crystal_SMD_HC49-SD_HandSoldering wastes too much board space

onitake, 2019-04-04:
The SMD variant of the HC-49 package has pads that are twice as large as necessary.
This wastes enormous board space and is rarely useful.

If a pads that are also compatible with larger packages are required, a different footprint should be created.

According to the data sheet http://cdn-reichelt.de/documents/datenblatt/B400/xxx-HC49-SMD.pdf , the pads should be 2x4.5mm, but they are 2x7.875mm. This is way too much, even for hand soldering. I built boards with this footprint, and 4.5mm pads would be perfectly sufficient.

evanshultz, 2019-04-04:
I think this might be just a case of footprint selection.

https://github.com/KiCad/kicad-footprints/blob/master/Crystal.pretty/Crystal_SMD_HC49-SD.kicad_mod, which references the datasheet you posted, is the standard crystal footprint. It has 4.5x2mm pad spaced 9.5mm apart, exactly like the datasheet.

https://github.com/KiCad/kicad-footprints/blob/master/Crystal.pretty/Crystal_SMD_HC49-SD_HandSoldering.kicad_mod, with a clear `_HandSoldering` suffix, has pads as you described above. The pads are again 4mm apart.

So you must be using the hand soldering footprint when you don't want to. Simply choose the standard footprint and it will have pads exactly as you wanted above. CvPcb may cut off the right side of the footprint name making it easy to confuse the footprints, so just be sure to pick the standard footprint.

I'm going to close this case since it seems like the footprint is fine and correct, but you can re-open if there is more to it that I didn't catch.

onitake, 2019-04-04:
My point was that even for hand soldering, 3mm on each side is wasteful.

I can use the regular footprint, yes, but that one leaves only a narrow space for heating the pad. The data sheet specifies 13mm edge to edge (4.5+4+4.5mm), and that just leaves a tiny 0.5mm space on each side for soldering.

So, both options are somewhat bad.

Is there a specific reason why the pads are so long? Perhaps an unwritten rule that hand-solderable pads need to be 3mm larger than necessary?

evanshultz, 2019-04-04:
It sounds to me like this is subjective. You are hand soldering and the pads are longer than you like. Somebody else, perhaps with a different skill set, different soldering iron tip, different solder, etc. may need longer pads and they created the current _HandSoldering footprint to suit them.

There is no guideline for hand soldering footprints that I've seen. I posted some colloquial thoughts at https://github.com/KiCad/kicad-footprints/issues/1302, and it seems similar to your suggestion in that much smaller pads are still reasonable.

Would you like to submit a few footprints with smaller pads? You could replace _HandSoldering with _Pad5.5x2 or _Pad6x2 depending on the pad size you think would work best. Either of those sound OK to me and I would be comfortable soldering it.

onitake, 2019-04-07:
I can do that, of course. However, I still think the _Handsoldering variant should be changed.

As soon as I have an actual photo that demonstrates this, I'd be happy to share it.
In the meantime, here's a comparison of of both existing footprints to illustrate:

Default variant:
[image]

_Handsoldering variant:
[image]

The 3D part was missing in KiCAD, I designed one myself and submitted a PR: https://github.com/KiCad/kicad-packages3D/pull/539

evanshultz, 2019-04-08:
Note to revisit changing the pads closer to v6's release.

antoniovazquezblanco, 2019-04-15:
What is the show stopper that doesn't allow us to target 5.1.2?

Thanks!