KiCad/kicad-footprints #1536: Texas S-PWSON-N10 issue

myfreescalewebpage, 2019-04-07:
Wanted to submit a new symbol, I have just checked if the footprint already exist. It is in the library but has not the right dimensions.

Existing footprint: Package_SON.pretty/Texas_S-PWSON-N10.kicad_mod
Dimensions: 3x3mm
Was added for: http://www.ti.com/lit/ds/symlink/tps63060.pdf

The device I want to submit: http://www.ti.com/lit/ds/symlink/tps54541.pdf
It has the same package name but it is 4x4mm.

Seems Texas package naming is not so clear and will need to pay attention for future submissions.... How is it possible to handle this for my new device ?

Thanks,
Joel

myfreescalewebpage, 2019-04-08:
Proposition: keeping package size and pitch every time ? In this case:
- TPS63060: Texas_S-PWSON-N10_3x3mm_P0.5mm
- TPS54541: Texas_S-PWSON-N10_4x4mm_P0.8mm
Joel

poeschlr, 2019-04-08:
In that case we can consider to completely drop the ti name alltogether as it then no longer realy adds anything (It can still be included in the description and keyword fields meaning a search will bring up the correct package)

We should however ensure that we do not change too many footprint names during a stable release phase. I might need to think about this a bit this weekend.

myfreescalewebpage, 2019-04-08:
According to me keeping Texas is great to explain from where the "S-PWSON-N10" comes from, but I'm not absolutely attached to it.

Also, I was thinking that we have probably several verification to re-check to ensure the symbols are correctly associated to the footprints because this kind of can clearly be forgotten during symbol reviews...

Sure, this is something to think, take the time required.

In my opinion the v6 is the opportunity to do this kind of modifications. People who want a stable release install and use v5.1 and do not follow the github repos. I have already discuss of that with @evanshultz because I think there is several huge enhancements like that to think and it is not possible to handle them just before the v6 release, we have to do it in advance.

evanshultz, 2019-04-09:
As I noted somewhere else, I think about a RF IC, TI told me the three letter name is the one to use. It is most descriptive and informative. You will see that play out again here where the footprint name shown above is the same but the three letter designation is different: DSC vs DPR.

In addition, the Texas_ prefix is valuable if the footprint isn't derived from a standard (like IPC) but instead adheres to the TI footprint given. In that case, we need the manufacturer prefix to indicate it's not a generic footprint but one meeting a specific manufacturer's recommendation.

evanshultz, 2019-04-09:
Found it. https://github.com/KiCad/kicad-footprints/issues/606

myfreescalewebpage, 2019-04-09:
Well done @evanshultz that's clearly the same issue applied to other devices :)

myfreescalewebpage, 2019-04-18:
@poeschlr any news for this subject ? Thanks for your return, Joel

poeschlr, 2019-04-19:
I think in future we simply go with the generic naming instead of using manufacturer specific conventions. (Unless it is a very specialized part.)

myfreescalewebpage, 2019-04-19:
Thanks. For the referenced footprint in my first message, can you give an example of how they should be named ?

myfreescalewebpage, 2019-05-30:
@poeschlr any help for my previous comment here ? Thanks so much, Joel

slowcoder, 2019-09-25:
Just a heads up. The PWSON-N10 footprint still doesn't work for TPS54541.  I found out the hard way.