KiCad/kicad-footprints #1555: Footprints description contains Device Datasheet

jstjst, 2019-04-21:
I noticed, that many footprint description fields contain the device datasheet and not the footprint datasheets form the manufacturer. This can be seen at manufactor specific footprints and general footprints:
https://github.com/KiCad/kicad-footprints/blob/master/Package_SO.pretty/SOIC-8-1EP_3.9x4.9mm_P1.27mm_EP2.41x3.81mm.kicad_mod

This makes it harder to find footprint. I think, if avalible, the footprint

evanshultz, 2019-04-22:
Hi Jonas,

There is some method to this. If the footprint is generated from a script, like many of the IC and connector footprints are now, the package drawing is the input to the script and the footprint a vendor supplies does not apply to our footprint. So in this case including the package drawing is correct. The first footprint you linked falls into this category and the description clearly shows it's a scripted footprint.

In other cases the footprints were handmade and the vendor footprint should be included. Many odd parts aren't scripted and so there will always be lots of parts in our library which do need the vendor-supplied footprint as a reference. In those cases, the documentation linked should show the footprint.

That all being said, I'm sure there are plenty of footprints that need some corrections and could be improved for all users by updating the description link.

jstjst, 2019-04-23:
I'm not sure if I understand you correct:

You stated, that the right document is link for the first footprint? The first footprint is not vendor specific, it is a commonly used footprint, so no vendor footprint code is given. But it has a device datasheet with a footprint drawing linked: https://www.analog.com/media/en/technical-documentation/data-sheets/ada4898-1_4898-2.pdf
But I think it would be possible and easy to find the drawing if we only link the footprint drawing: https://www.analog.com/media/en/package-pcb-resources/package/pkg_pdf/soic_edrd/rd_8_1.pdf

For the second footprint, there is the wrong document linked?

evanshultz, 2019-04-23:
OK. Let's look specifically at the two you showed above instead of generalizing.

For the first footprint, we can see that it's scripted and so only a package drawing is needed. That package drawing is on page 17 (the link to page 29 is wrong) and it even indicates compliance with JEDEC MS-012. The script then should use JEDEC MS-012 dimensions and mention this industry-standard package in the description. However, since the JEDEC docs require registration to get them directly from JEDEC we chose datasheets that shows compliance but were freely-accessible. Just linking to the package drawing and not a datasheet would be my preference as well, but the package drawing is the same either way so we'll consider both acceptable. There is no recommended footprint in this datasheet.

For the second one, we now know the three-letter code from TI is the best differentiator for naming footprints. See https://github.com/KiCad/kicad-footprints/issues/1536. So this footprint really would be best named as `Texas_DRC` or something like that. However, updating the footprint names requires updating symbols and nobody has put in the effort to do that. Yet. Nonetheless, if just a package drawing could be found that would be nice as well. And while it seems the documentation is OK (I didn't verify it against the footprint myself) I agree that the package name `DRC` or `DRC0010J` should be in the description and/or keywords.

Does that help to clarify things?

jstjst, 2019-04-23:
Yes, good explanation

closed 2019-04-30 by jstjst