KiCad/kicad-footprints #1556: SO vs SOIC

evanshultz, 2019-04-22:
We have a number of footprints which are prefixed with 'SO':
[image]

I mentioned one that I don't believe is needed at at https://github.com/KiCad/kicad-footprints/pull/1333#issuecomment-485519199.

https://github.com/KiCad/kicad-footprints/blob/master/Package_SO.pretty/SO-20_12.8x7.5mm_P1.27mm.kicad_mod seems like it could easily be SOIC and fit in well with the other footprints.

But what about the others? That leaves just 4, 5, and 6 pin footprints starting with 'SO'. Do we have any guidelines about how, when, or if to drop the 'IC'? Should they all be 'SOIC' since the vast majority of P1.27 and P2.54 footprints are this way?

myfreescalewebpage, 2019-04-22:
@evanshultz my personal position is that the footprint associated to a symbol should follow the content of the datasheet. This prevent future questions.

calebreister, 2019-04-22:
It is worth noting that the SO and SOIC footprints are not always identical, and the definition may very based on the manufacturer. For example, the [LT1118](https://www.analog.com/media/en/technical-documentation/data-sheets/1118fd.pdf) comes in an "SO-8" package. However, if you look at the bottom of the datasheet, you will find that the package dimensions are 3.9x4.9mm, which matches SOIC-8 in the KiCad library. Although both packages have the same pitch, SO-8 has a larger outline (5.3x6.2mm).

Note: the LT1118 is not in the official KiCad library (*yet*). I'm using it in a project, and plan to contribute it when I get the chance to make sure it is fully KLC-compliant.

myfreescalewebpage, 2019-04-23:
Good example from @calebreister !

evanshultz, 2019-04-23:
The nomenclature used by a specific manufacturer should not matter. What if another company reversed the terms in their documentation? What does that mean to us?

Thanks to Rene, we have a nice gullwing footprint generator that uses IPC rules and DOES NOT rely on what any individual vendor says for industry-standard package types. So I think internal consistency is the important thing to consider.

To close the loop, LT's "SO-8" package is fully compatible with [JEDEC MS-012](https://www.jedec.org/taxonomy/term/2343), which uses only "Small Outline" in the document body but falls under the "SOIC" category on their website. IPC-7351B also does the same thing:
[image]

So... I think using SOIC for a generic footprint is the right way to go.

myfreescalewebpage, 2019-04-24:
Hummm but how to know if "SO" package for a specific device of a specific manufacturer can fit the SOIC definition ?

evanshultz, 2019-04-24:
Body size and lead pitch are part of the footprint name.

myfreescalewebpage, 2019-04-24:
Sure.