KiCad/kicad-footprints #1559: Using git hooks to run tests before committing

5N44P, 2019-04-26:
Hi guys!
I'm relatively new to git, and today i've learned the existance of the git hooks, letting you run a script before committing. 

It would be intresting to run the KLC helper script for footprints [(check_kicad_mod.py)](https://github.com/KiCad/kicad-library-utils/blob/master/pcb/check_kicad_mod.py) before making a commit (on the local branch, before pushing to the contributor's remote fork, before opening the PR), aborting the commit if some error is given by the script. In case of known warnings that can be ignored, it seems that `--no-verify` can be used to skip the git hooks checks.

+ What are your thoughts on this? Have i just discovered the wheel? 
+ Has anyone implemented a pre-commit hook for this and can help me implement it on my client? 
+ Is it just a bad idea?

The same could probably apply to the sch symbols and the 3D models, but i really don't know since i've never contributed anything there.

poeschlr, 2019-04-27:
Not really a topic where we have any experience. If you want to know this in general then stack-exchange or a similar software development forum might be the best place to ask for support.

I don't think we have any check scripts for 3d models right now. I am not even sure what we could reliably check without considerable development effort.

cpresser, 2019-08-23:
Indeed, it would be nice to have the output of the helper-scripts before commiting. But I agree with @poeschlr, there are to many false positives.
It seems like you already figured on how to run the scripts manually in your local tree. 

The contribution guidelines (http://kicad-pcb.org/libraries/contribute/) already cover this. 
Do you think it would be a improvement to add a new point between 5 and 6?
> 5a: (optional) use the helper-scripts to check your new changes for common problems. Be aware that the script-output is just a helper. You need to check each reported violation manually.

chschlue, 2019-08-23:
This is nice, but is it worth the effort?
I would image people regularly forgetting to update their local scripts and then missing errors after the commit thinking they already checked for themselves, only complicating things.
PRs can always be (and are usually?) squash merged, so spamming commits until a PR passes the tests doesn't hurt much.
Just my two cents.