KiCad/kicad-footprints #1560: [WIP] Xilinix BGA footprints

herostrat, 2019-04-26:
This is a tracking issue to compile the work of footprints for Xilinix BGAs.

**1. Artix-7**
 - [ ] Script PR: https://github.com/pointhi/kicad-footprint-generator/pull/343
 - [ ] Footprint PR: https://github.com/KiCad/kicad-footprints/pull/1557

**2. Spartan-7**
 - [ ] Script PR: 
 - [ ] Footprint PR:
_Other Initiatives:_
Script PR: https://github.com/pointhi/kicad-footprint-generator/pull/344
Footprint PR: 

**3. Kintex-7**
 - [ ] Script PR: 
 - [ ] Footprint PR:

**4. Virtex-7**
 - [ ] Script PR: 
 - [ ] Footprint PR:

**5. Zync**
 - [ ] Script PR: 
 - [ ] Footprint PR:
_Other Initiatives:_
Script PR: https://github.com/pointhi/kicad-footprint-generator/pull/240
Footprint PR: https://github.com/KiCad/kicad-footprints/pull/1167

Additional Work:
 - [ ] Remove old and/or wrong packages that reference to Xilinix FPGAs

evanshultz, 2019-05-14:
Thanks so much!

Is there any more renaming that needs to take place to ensure all applicable package names are captured in the footprint names?

herostrat, 2019-05-14:
Afaik those were all BGA packages Xilinx uses for the 7-series chips, so no, I don't think so.

The remaining work is cleaning up the "old" packages (as you previously stated according to https://github.com/KiCad/kicad-footprints/pull/1167#issuecomment-447585654) and close similar PRs.

evanshultz, 2019-05-14:
@herostrat 
https://github.com/KiCad/kicad-footprints/pull/1167 and https://github.com/pointhi/kicad-footprint-generator/pull/240 have a section named "Duplication of existing parts". Are you open to checking that out to see if those are footprints that are unneeded? It appears they are not covered in the list above.

herostrat, 2019-05-14:
Yes, I will, this will be the next workpackages for me.
The ones listed above are just the starting point I already identified.