KiCad/kicad-footprints #1561: molex connectors only go up to 15

mxa, 2019-04-27:
I am looking for the 16 pin molex connector [52746-1671](https://www.molex.com/pdm_docs/sd/527461671_sd.pdf) but can't find it.
The CLIK-Mate Series looks similar but only go up to 15 pins.

poeschlr, 2019-04-27:
The part you link is an FFC/FPC connector, click mate is completely different.

We do not have the 52746 series at all in the lib. (Meaning i interpret your thing as a request for making footprints. We do not take such requests which means this is invalid as an issue as far as i am concerned and will be closed for that reason.)

---

If you are interested to make footprints for this series and want to add it to the lib then i suggest you look into our scripts. With a bit of luck the generic connector script can be directly used or easily adapted for this part: https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts/Connector/Connector_SMD_single_row_plus_mounting_pad (I am not sure if it supports the fact that the clamping part is wider than the main body. So you might need to write a specialized script for this series. For that you would still start with a similar script.)

closed 2019-04-27 by poeschlr

mxa, 2019-04-27:
meanwhile i made one: [molex 52746-1671.kicad_mod.zip](https://github.com/KiCad/kicad-footprints/files/3124259/molex.52746-1671.kicad_mod.zip)