KiCad/kicad-footprints #1583: Potentiometer_Omeg_PC16BU_Vertical wrong pad number

McColson, 2019-05-07:
Hello,

I think the footprint of 'Potentiometer_Omeg_PC16BU_Vertical' has a problem with the pad number.
In fact, the number 1 and 3 are interverted.
The Silkscreen is on the front layer, so the pin 3 should be at the bottom of the picture.
[image]

Matthieu COLSON

evanshultz, 2019-05-07:
I believe this is covered by https://github.com/KiCad/kicad-footprints/issues/188. Please confirm and, if so, close this issue. I would like to see this all resolved for v6, but since it will (may?) be a breaking change it won't happen in the near future.

McColson, 2019-05-08:
Hi,
I'm not totally sure to understand well the #188 but, in the picture here : https://user-images.githubusercontent.com/1936989/35872082-bc1a3ae4-0b1a-11e8-8780-868674a6d77e.png
The ALPS at the left of the omeg footprint is good, the pin numbers are correct (compare to the symbol of a potentiometer in a Kicad's schematic), the omeg isn't. Maybe it's that you call the CW or CCW markings problem ?

evanshultz, 2019-05-08:
To make generic symbols the pin numbers of each individual part aren't considered.

That being said, the pin numbers aren't consistent or namely intuitively in KiCad at this time.

McColson, 2019-05-08:
Ok thanks you for your reply. I close the issue.

closed 2019-05-08 by McColson