KiCad/kicad-footprints #1589: Holes wrong in OSROM relays

hildogjr, 2019-05-09:
I did the pull requests #1587 and#1588 for fix the hole error in the OSROM realys, see datasheet https://omronfs.omron.com/en_US/ecb/products/pdf/en-g5q.pdf

But I am getting a CI error of layer overlap that I don't understand.

evanshultz, 2019-05-09:
Well, you need to look at the footprint:
[image]

Here is the exact error message:
```
Violating F5.4
    Elements on the graphic layer should not overlap
    F.SilkS graphic elements should not overlap.
     - The following elements do overlap at least one other graphic element on the same layer
     - Line (2.03,-4.83) -> (2.54,-4.83) on layer 'F.SilkS'
     - Line (3.05,-2.29) -> (2.54,-2.29) on layer 'F.SilkS'
```

So a silk line is overlapping something on the same layer. (I'm not familiar with this exact part of the check script but it seems clear so I continued on.)

Looking at the above and checking the given locations, it appears the error is because the top and bottom of the coil are composed of two horizontal lines. Delete one and extend the other so the rectangle indicating the coil just uses four line and all will probably be well.

(Since only the endpoints are in common this error might be a bit aggressive, but if it's only checking for lines in the same orientation where they can be a single line it's relatively valid.)

IMO it's not required to make other updates when correcting a true issue, so I will merge without improving the silk lines. But if you want to tackle that fix I'm happy to merge those improvements too.

The silk outline is offset quite far from the fab outline. You could shrink the offset to 0.11-0.2mm if you like while you're at it. But again, this is not required.

hildogjr, 2019-05-09:
Thanks. Maybe I will deal with other time, because appear that the checking is complain of overlap because the beginning-end of different line at same coordinates.

evanshultz, 2019-05-09:
That's fine. I merged both of the PRs you submitted to fix the hole size issue. Thanks again!

closed 2019-12-10 by evanshultz