KiCad/kicad-footprints #1645: Common mode choke location

evanshultz, 2019-06-13:
We have CMC footprints named in various and inconsistent ways, and I'm not sure that they belong in an inductor library since their purpose is rather different than just an inductor:
- Inductor_SMD: L_CommonMode_Wuerth_WE-SL2 (qty=1)
- Inductor_THT: Choke_EPCOS_B82722A (qty=1)
- Inductor_THT: Choke_Schaffner_RN* (qty=16)
- Inductor_THT: L_CommonMode_Toroid_Vertical* (qty=10, and actually Bourns 81xx-RC)
- Inductor_THT: L_CommonMode_Wuerth_WE-CMB* (qty=6)

I'm thinking they should be grouped together. Perhaps they all start with `L_CommonModeChoke` or just `L_CommonMode`?

And it seems like there's enough to populate a new footprint library. Perhaps `CommonModeChoke` or `L_CommonMode` (I suppose we could be split by SMD and THT if desired)?

Anybody else have feelings on this?

cpresser, 2019-08-12:
I agree, they deserve a different library.
I think "L_CommonModeChoke" is a good name, I would not shorten it to "L_CommonMode".

Will this be a v6 thing? Since moving footprints will break existing designs?

evanshultz, 2019-09-10:
Moving footprints should be OK since footprints are embedded into the board file. Rene has been allowing footprint changes at any time.

v6 should do a similar thing for symbols so we may get the same flexibility there too.

But regardless, we should use common sense and not move things around all the time for little benefit. IMO, even if it doesn't break designs, this isn't worth doing now and can wait for v6 or shortly before.

poeschlr, 2019-09-10:
I advice against making new libs even for the footprint side as users lib tables do not update. (At least not new libs for old footprints. Meaning do not split libs during a minor release)