KiCad/kicad-footprints #1654: Connector footprint library organization

evanshultz, 2019-06-20:
There is a bit of a problem with connector libraries: It's sometimes hard to figure out where a particular footprint goes!

Let's take an example of a single connector that is from a vendor with their own library. Does this new footprint go in:
1. Connector.pretty
1. Connector_<vendor>.pretty
1. Connector_<function>.pretty

An argument could probably be made for all three, but the latter two are where I run into the most questions. How to decide? Is there an overall priority where we go with the connector function (HDMI, USB, etc.) if a library exists? So function overrides vendor? Or has this never been discussed before?

This came to my mind when looking at https://github.com/KiCad/kicad-footprints/pull/1636. They are FFC connectors, but they're from Molex.

poeschlr, 2019-06-21:
This is actually clarified in the KLC http://kicad-pcb.org/libraries/klc/F1.2/

My intention when making the original draft (that got converted to this rule by oliver) was the following:

There are two types of connectors. Ones where the mating end is standardized (USB, DSUB, FFC, ...) and ones where it is a manufacturers standard (Phoenix MSTB, JST XH, ...)
The former are organized by the standard (So all USB connectors live in the USB lib.) The later are organized by manufacturer.
Both types of libs can be further split up. (The Phoenix libs are examples of that.)

The Connector lib is there to collect footprints for parts that do not fit anywhere else but there are simply not enough similar parts to merit making one.

evanshultz, 2019-06-21:
Alright. Thank you.

Let me add some more details just for my own providence.

A USB connector can't go into `Connector` but must go into `Connector_USB`. If there are a lot of Molex USB connectors, they may move into `Connector_USB_Molex`. `Connector_USB` would then only contain USB connectors not made by Molex. And `Connector_Molex` would never contain any USB connectors.

And all that means that the PR I referenced went into the correct lib. Yay!

closed 2019-06-21 by evanshultz