KiCad/kicad-footprints #1663: Connectors with wrong orientation in footprint name

evanshultz, 2019-06-25:
There are a number of connector footprints with the wrong orientation because of a typo in the scripts. This issue is to track them so for v6 the old ones can be removed and new, corrected ones can be added.

- [ ] Hirose DF63 (new ones are pushed already so only `_Horizontal` ones need to be removed)

poeschlr, 2019-06-26:
We are allowed to rename footprints. The reasoning is that they are embedded in designs and do not rely on the temperamental rescue feature. (Which was the reason given by @stambaughw for not changing symbol libs too much during a stable release.)

Also: if a wrong name can mean users get confused about which part it is then it would be classified as a bug and can even be changed for symbols. (A typo in the part number would be a typical example in the symbol side. the wrong cleartext naming of a footprint another. Even if the part number is included as well as users will not really parse that as easily.)

antoniovazquezblanco, 2019-07-09:
Added some changes to fix this to https://github.com/pointhi/kicad-footprint-generator/pull/392 I will now generate the corresponding footprints.

If this is too much I can revert the changes.

Thanks!

evanshultz, 2019-07-09:
This should add duplicate footprints (`_Horizontal` suffix for horizontal footprints that were incorrectly suffixed with `_Vertical` before). I understand we are allowed to change/remove footprints at any time since they're sucked into the board file.

I started a list above about which connector series have dual suffixes and which suffix is to be removed. Can you please add to the list?

antoniovazquezblanco, 2019-07-09:
Ohh! I though that this was to keep track for removal during v6 but was finally moved to v5.3. I've added the removal of the old DF63 footprints to my PR and checked all the Hirose connectors. That PR should tick all the `Connectors_Hirose` footprints when merged. But I do not have time to go all over all the connectors available.

evanshultz, 2019-07-09:
Yeah... well... I still feel nervous to delete footprints at will but I understand Rene is saying that's OK. I don't think waiting until v6 is a big deal, but if this can be done sooner with Rene and the dev team understanding and approving then so be it.

antoniovazquezblanco, 2019-07-10:
I am also nervous about it but footprints will not disappear from your board if no longer in the library and poorly named footprints can lead to confusion and errors when designing new boards.

Footprint renaming has been done in previous stable releases (e.g. v5.1.1) without any complaining from the devs.

poeschlr, 2019-07-12:
Footprints can be renamed without issue. I would still require good reason to do so as to keep the library from being too volatile. (Something like this is a clear case where the name change can be easily explained. Changing naming conventions however is something that is harder to explain and should be avoided even across major releases if there is no tangible benefit to the change.)

antoniovazquezblanco, 2019-07-16:
Fixed

evanshultz, 2019-07-16:
I'm not sure if there are others, but just clicking through the generic pin and manufacturer-specific connector libraries I found two other (related) libraries which appear to have vertical and horizontal confused.