KiCad/kicad-footprints #1674: SD_Connector: wrong footprint orientation

gmsotavio, 2019-07-06:
During my inspection, I've found which several footprints in SD_Connector library have a wrong orientation. KLC F4.2 states: "Footprints should be oriented such that Pin 1 is located in the upper left corner (IPC-7351)." (http://kicad-pcb.org/libraries/klc/F4.2/). Am I right?

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

evanshultz, 2019-07-08:
There are many footprints which are older and do not adhere to the latest KLC guidelines. I think these fall into that category. Thanks for pointing out these issues so they can be resolved.

If time permits, I understand Rene allows footprints to be changed at any time. However, it would certainly be nice to get this done for v6 so I've added that milestone.

antoniovazquezblanco, 2019-07-09:
@gmsotavio will you be willing to open a PR to fix the issue?

Thanks

gmsotavio, 2019-07-10:
@antoniovazquezblanco OK. Now that the issue has been confirmed, I will provide the fix. Thanks.

cpresser, 2019-09-06:
Related question: Should we apply KLC 4.2 for all connectors? There are quite a lot (non SD-Card connectors) that don't follow that rule.
For example, I just looked at https://github.com/KiCad/kicad-footprints/pull/1770. 
Or "MMCX_Molex_73415-0961_Horizontal_1.0mm-PCB".
[image]

evanshultz, 2019-09-06:
Yes, we should. Pin 1 should normally be in the top left. It's on the left for 2-terminal footprints (not sure this applies to any connectors, but think about chip resistors). And in the top center if pin 1 isn't on a corner (like PLCC footprints).

Since this is such a minor change, I'd suggest making an issue, marking it for v6, and creating a list of connector [families] which need updating. This can be queued and then implemented closer to v6's release date. No need to make branch conflicts or change things for only a minor KLC fix right now.

poeschlr, 2019-09-07:
Most of our connector footprints are scripted and should full fill this rule already. So i guess it will be only a few that are problematic. I agree that the first step would be assessing what needs to be done and that this would be best in an issue. After we have this information we can then decide if it makes sense to fix it now or if it is best left for later.