KiCad/kicad-footprints #1712: Where to put connector mating information

calebreister, 2019-07-16:
I'm working on adding a large portion of the [Samtec Q Strip](https://www.samtec.com/connectors/high-speed-board-to-board/mezzanine/qstrip) connector family to the KiCad libraries, and have written a [wizard](https://github.com/calebreister/kicad-footprint-wizards/blob/CalebReister/QStrip_wizard.py) (a more automated version is in the works) capable of generating both terminals and sockets. Within this family. For any given connector in the family, there are likely two or three sockets that can mate with it. I have several questions:

- Should a list of compatible connectors be provided in the footprint metadata?
- If so, where should it be specified? Should a `mate` property (or something similar) be added?
- Should the list include all connectors that are *mechanically* compatible (not necessarily pin-compatible)? Example: single-ended terminals can fit in differential sockets, but some of the pins are unused.
- Should compatible connectors that are not in the library be included in the list?

evanshultz, 2019-07-17:
This is my opinion as just one of the librarians: I wouldn't list compatible connectors. It will rot in some fashion. If the documentation lists mating connectors or both genders are in the datasheet that should be relatively easy to find.

As these connectors are for a specific standard I expect folks who need this connector to be versed in it. If you mention the standard to which these connectors adhere I think we'll be good.

Also, the footprint generators the librarians use are at https://github.com/pointhi/kicad-footprint-generator. It appears at a glance that the repo you're working on is an old creation of the previous head librarian. (Yes, the repo locations make this confusing.) While we can accept the footprints with no problem, for maintainability it is much preferred to use the modern framework I linked above. Again you need not change anything you're doing this is just sharing information.

calebreister, 2019-07-17:
@evanshultz Ok, I'll leave that information out. I'm currently reworking my wizard code using the newer framework. Note that the vast majority of Q Strip family are general purpose high-speed connectors, of which the HSMC card connectors are a special case.

calebreister, 2019-07-23:
Update: here is the PR for the newer script.

https://github.com/pointhi/kicad-footprint-generator/pull/397

closed 2020-09-06 by calebreister