KiCad/kicad-footprints #1756: Taiyo Yuden SMD inductor solder pads bigger than MF recommended

seppestas, 2019-07-29:
The L_Taiyo-Yuden_NR-xxxx footprints do not seem to follow the manufacturer recommended landing patterns. All pads seem to be 0.2mm longer (Y axis) than recommended.

E.g, the length of the pads of the L_Taiyo-Yuden_NR-40xx footprint is 3.9mm in [this libraries KiCad module](https://github.com/KiCad/kicad-footprints/blob/cf71819ab37168bfad298de859b1bbd00f5a93f6/Inductor_SMD.pretty/L_Taiyo-Yuden_NR-40xx.kicad_mod), but it is only [supposed to be 3.7m according to the datasheet](https://www.yuden.co.jp/productdata/catalog/wound04_e.pdf).

Is this a mistake, or was this done to improve solderability?

seppestas, 2019-07-29:
@SchrodingersGat it looks like [you generated these footprints "as per datasheet" in the original Inductors.pretty](https://github.com/KiCad/Inductors.pretty/commit/70424128869bbb13c27fcbccf8b3db344ec662a7).

Could you clarify if this is a mistake or intentional? If intentional, is there a rule for it?

evanshultz, 2019-07-29:
@seppestas 
Thanks for bringing this up. That means it's an old footprint and subject to loose interpretation.

Making bigger pads could be helpful in some circumstances, but it should not be the expectation in this library if the footprint name or description don't indicate any deviation. We'll wait for @SchrodingersGat  to respond in case he can shed any light on this, but I'm of the mind the footprints should be updated to match the datasheet.

And note that the documentation linked in the original PR is not available. The parts do not match the _current_ datasheet, but that doesn't mean they weren't built correctly according to an older version of the datasheet.

Either way, I'm marking this as a bug.

SchrodingersGat, 2019-07-30:
Hi guys, this was a fair while ago so I cannot specifically remember why this deviation was made. Sounds like a PR should be made to fix this.

PR #1792 (2020-10-01): Fix pad sizes of Taiyo Yuden NR series SMD inductors
Fixes #1756 for Taiyo Yuden NR series SMD inductors by correcting the pad width so it matches the manufacturer recommended pad width.

Also simplifies the outline on the F.Fab layer. For some reason the "sides" where split in 2:
[image]

Note that this does not change the "HandSoldering" footprints. I assume the pads for handsoldering are intentionally bigger and I suspect the wider pad size was accidentally copied to the non-handsoldering components.
closed 2020-10-01 by evanshultz