KiCad/kicad-footprints #1757: USB Type C Receptacle

G-Pereira, 2019-07-30:
Can someone make a footprint for this?
[image]

cpresser, 2019-07-31:
Usually requests don't work.
Anyway, In order that someone actually want to tackle this, can you provide a name and datasheet of that connector?

poeschlr, 2019-07-31:
We are sadly unable to take requests as @cpresser already indicated. If you need support then i suggest the forum https://forum.kicad.info/. Its FAQ section might be a good read for you. This article in particular: https://forum.kicad.info/t/tutorial-how-to-make-a-footprint-from-scratch/11092

closed 2019-07-31 by poeschlr

G-Pereira, 2019-08-05:
I had never worked with the footprint editor but I found out that it is easier than I thought. I am working on this.

calebreister, 2019-08-05:
@G-Pereira Just make sure it adheres to the [KLC](http://www.kicad-pcb.org/libraries/klc/) if you want to contribute it. Also, take a look at the [contributing guidelines](http://kicad-pcb.org/libraries/contribute/).

cpresser, 2019-08-05:
Could this be that connector? They look very similar: https://github.com/KiCad/kicad-footprints/pull/1631

G-Pereira, 2019-08-07:
> @G-Pereira Just make sure it adheres to the [KLC](http://www.kicad-pcb.org/libraries/klc/) if you want to contribute it. Also, take a look at the [contributing guidelines](http://kicad-pcb.org/libraries/contribute/).

Thanks, I am aware of KLC since I already contributed to several symbols. Although I am not familiar with the footprint part of KLC, I will be sure to check it carefully.

> Could this be that connector? They look very similar: #1631

Sadly no. I have made an extensive search on the library and here on github before starting to create the footprint myself although I found very strange because it's one of the most popular items on AliExpress. That makes me think that my work will definitely help others as well. :)