KiCad/kicad-footprints #1764: ST_VL53L1X wrong rotation?

michael-p, 2019-08-03:
Dear all,

I think the footprint OptoDevice/ST_VL53L1X is rotated incorrectly by 180 degrees. The part was added in #1623.
Looking at the [datasheet](https://www.st.com/resource/en/datasheet/vl53l1x.pdf), Figure 19, you can see that Pin 1 (which has this little "nose") should be in the bottom right corner, however, in the Kicad footprint it is in the upper left corner (see #1621 for an image). Note that Figure 18 and 19 in the datasheet are quite confusing because they switch between "looking from top", "looking from bottom", and "looking through top" views without clearly stating it!

Despite the rotation-issue the footprint seems to be fine, so it's generally useable but will cause problems with pick-and-place machines.

I did not yet check the other part that was added in the merge request (VL530X).

Best
Michael

cc @5N44P @poeschlr

5N44P, 2019-08-03:
Hi there, 
The rotation is imposed by the Kicad Library Convention, see [KLC F4.2](http://kicad-pcb.org/libraries/klc/F4.2/), which states: 

> Footprints should be oriented such that Pin 1 is located in the upper left corner (IPC-7351).

michael-p, 2019-08-03:
Thanks for the info, I was not aware of this convention! Do you have any advice on how to handle the rotation then? I was assuming that the footprint has to be drawn with exactly the same orientation as in the datasheet so that pick & place machines can use the correct orientation (relative to the tape). In fact I just got a board back from Macrofab where this part is off by 180 degrees :)

Kind regards
Michael

cpresser, 2019-08-05:
Hi Michael,

I feel your pain, I had a similar issue recently.
But I agree that the footprint is correct. The Pin1 Marker is correct in F.Fab and F.Slik layer.
The CAM-Operator is supposed to check the datasheet, it contains the info where Pin1 is. That can be checked against F.Fab.

[image]

One possible improvement of the footprint would be to add information about the protective liner. That could be added to the Cmts.User layer.

evanshultz, 2019-08-05:
IPC 7351 has always used pin 1 in the top left where it works (it does there, and there are other conventions for different part shapes) and so KLC adopted that industry-standard convention. (We can ignore later revisions to that document which support other, optional, component orientations.)

The datasheet is never to be understood as representing the component's orientation in the packaging. Many parts are available in various packaging and the datasheet need only show the package. Note package and packaging are different.

Indicating anything about the reel in the footprint is likely only adding pain later. What is other packaging becomes available? What if different reel sizes use different package orientation? What is the SMD machine only supports trays and so the parts are re-packaged. A general-purpose library like this one can't know all of those downstream details and would cause as many problems as they might potentially clarify.

In my experience at various assembly houses of various sizes in multiple continents, this is not required to be part of the footprint. So I'm dubious if it would really help anyone or just make contributions more difficult for no benefit. I don't know every situation, of course, but I can't see an upside.

Placing parts at the wrong orientation is always a huge frustration and pain so I do understand the desire to figure this out and get it right. Should the footprint have been wrong it would have been something nice to resolve and I appreciate your effort, Michael, to raise up a potential problem.

I'm closing this issue since it seems resolved and I'm not aware of any further action needed.

closed 2019-08-05 by evanshultz