KiCad/kicad-footprints #1777: Where are the 3D models referred in the footprints?

DomoticX, 2019-08-11:
Hello, it seems some 3D shapes are missing?

for example in SOP-8_3.76x4.96mm_P1.27mm.kicad_mod

https://github.com/KiCad/kicad-footprints/blob/master/Package_SO.pretty/SOP-8_3.76x4.96mm_P1.27mm.kicad_mod

the link to shape is:
${KISYS3DMOD}/Package_SO.3dshapes/SOP-8_3.76x4.96mm_P1.27mm.wrl

https://github.com/KiCad/kicad-packages3D/tree/master/Package_SO.3dshapes

but does not exist....

SOP8 is common users there are only 4 and 16 pin in the libs??

am i missing something?

gmsotavio, 2019-08-11:
Hello @DomoticX,

You are right, the 3dshape model seems missing. The reason for this is there are no available 3D models for all footprints in current KiCad library. Make a footprint is easier than make a 3d model. Moreover, not all contributors have 3D modeling ability or even time to make a 3D model. The link you have mentioned is not automatically generated. Actually it is purposely inserted in the footprint, avoiding the need to make changes in the footprint if someone add the 3D model later. The footprint 3D model requirements is described in KLC F9.3 (http://kicad-pcb.org/libraries/klc/F9.3/)

Regards.

DomoticX, 2019-08-11:
Alright thanks!, ps. may i have one suggestion?, i also see some referral link to external pdf in the footprints, these pdf links decay over time (404 error and so...) why not put all the PDF docs offline in a folder and/or install with kicad?, the extra 500MB is not much (since kicad is 5-6GB) and the documentation is then always available (also offline) i can help you with that if needed?, what do you think?

poeschlr, 2019-08-12:
Regarding docs: It is not because of memory but because of maintenance effort. If the link fails then the user is at least aware of the fact that the info is out of date. With a local pdf there is no chance to ever discover that. Meaning local pdfs would require regular checking which is simply not possible with our manpower.

And of course there is always the question of licensing and copyright.

poeschlr, 2019-08-12:
@gmsotavio you seem interested in helping out with our effort. Would you be prepared to join the team?

closed 2019-08-12 by poeschlr

gmsotavio, 2019-08-16:
@poeschlr Sorry for the delayed reply. KiCad has been a very useful tool for me in the last years. Whenever I can, I use some of my spare time to contribute in some way. What I need to know before joining the team? Thank you for the invitation.
Regards

DomoticX, 2019-08-16:
@poechlr, The info in the PDF does not go out of date ...there could be some revisions, but that's more like fixing typo in the datasheets, however the chip could be obsolete within a decade, but doesn't mean it cannot be used then. 99% of every chipmaker provide the datasheet PDF's for free, how else would they sell chips without describing the functions of them? ;-), i sure can help here and there collecting data if needed...

chschlue, 2019-08-16:
> The info in the PDF does not go out of date ...there could be some revisions, but that's more like fixing typo in the datasheets,

There could be missing commas.
There could as well be wrong pinouts, wrong thermal characteristics, wrong (or out of date) packaging information, wrong whatever...

> 99% of every chipmaker provide the datasheet PDF's for free

99% of YouTube videos are provided for free. That does not necessarily mean you may redistribute them at will. And you may certainly not just stick a CC-BY-SA to them and distribute them with some random software package.

chschlue, 2019-08-16:
[image]
[image]
[image]

Pin functions, regulatory approvals, insulation characteristics, mechanical dimensions and other irrelevant nonsense.

gmsotavio, 2019-08-25:
As @poeschlr highlighted, there are licensing and copyright issues too. As an example, lets take a look in the datasheet of Keystone Test Point 5000 Series
[image]

The document states "THIS DOCUMENT IS THE PROPERTY OF KEYSTONE ELECTRONICS CORP. AND SHALL NOT BE REPRODUCED, **DISTRIBUTED** OR USED AS A BASIS FOR MANUFACTURE WITHOUT PRIOR WRITTEN PERMISSION FROM KEYSTONE ELECTRONICS CORP."

DomoticX, 2019-08-26:
Yes, some datasheets may have these writings but limited to a few, normally any chip datasheet isn't copyrighted (e.g. CD4066):

https://www.electroschematics.com/wp-content/uploads/2011/04/cd4066-datasheet.pdf

Ps. the cherrypicked datasheet with changelogs looks like taken from a cpu chip or PCB ;-) normally a "simple" chip doesn't have this large changelog...

jkriege2, 2019-08-26:
The situation is actually rather simple: all material on a manufacturers homepage belongs to that manufacturers (or any website owner for that matter) by default and unless stated otherwise! Also you don't need to explicitly state that a work is copyrighted. The author (or in our case often the employer) is automatically granted that right and with it the right to redistribute and relicense at will ( see e.g. https://en.m.wikipedia.org/wiki/Copyright). It is the same as with other content on e.g. a webpage: just because you're free to use it does not mean you're free to redistribute or reuse it or derive from it. 
In addition: many manufacturers state somewhere in their site (often in an impression and a bit hidden) a license that covers all their materials and that - in many cases - is similar to the example above, I.e. you may download and use the provided documents for their intended purpose, but you may not redistribute and relicense (which are the rights we would both require to put these datasheets into a github repo)

Hope that clears this up.
Jan

jkriege2, 2019-08-26:
Just some examples:
- https://www.microchip.com/about-us/legal-information

"Website Usage & Limitation of Liability: Use of any materials, information, text, graphics or data provided on Microchip’s Internet website is governed by our Terms of Use. Click here to read further."

https://www.analog.com/en/about-adi/landing-pages/001/terms_of_use.html

"ADI grants you a revocable, non-transferable, non-exclusive license to view, print out or download a single copy of the ADI Information, solely for internal non-commercial or informational use provided that you do not remove any copyright, trademark or other proprietary notices. Other than the foregoing limited single copy license, no transfer of any other rights is made or intended with respect to the ADI Information and the ADI Site. ADI may terminate this single copy license at any time for any reason, including for any breach of these terms of use, or breach of any other ADI agreement or policy. Upon termination you will immediately destroy all ADI Information in your possession or control."

... and you will find similar licenses/terms for all manufacturers!

poeschlr, 2019-08-29:
@gmsotavio we do not really have pre requirements for joining the team other than your willingness to do so (Or rather: If we ask then you already fulfill what we look for.) 

It is of course expected that you will need a bit of support by more experienced members at the beginning. I sent you an invite via github so if you are willing to dive in then just accept. (If you have any questions during a review then just ping us more experienced guys. You can also reach at least me and evan over at the forum.)

gmsotavio, 2019-08-30:
@poeschlr I accepted. It will be great to be part of the KiCad team. Since I started opening my pull requests here, I have learned a little about the workflow of librarians. But I think I'll still need some support in the beginning. Thank you for the invitation! ;)

poeschlr, 2019-08-30:
Just remember the golden rule "if in doubt ask" (and the second one "never merge something without testing it first" -> implies "never merge your own work") and you should be good to go.

chschlue, 2019-08-30:
> Ps. the cherrypicked datasheet with changelogs looks like taken from a cpu chip or PCB ;-) normally a "simple" chip doesn't have this large changelog...

These changelogs weren't exactly "cherry picked", I just happened to have them handy.
Also, they all belong to pretty simple ICs (20 pins max).

Either way, even if grave changes are not very common, they do happen often enough.