KiCad/kicad-footprints #1793: WiFi modules in both RF_WiFi and RF_Module libraries

gmsotavio, 2019-08-31:
Hi,

I noticed there is two libraries which have WiFi Modules. For example, the RF_Modules has the ESP-07. On the other hand, RF_WiFi has the USR-C322. Both them are Wi-Fi/RF modules. in my understanding, USR-C322 should be moved to RF_Modules. The RF_WiFi should be populated with WiFi ICs, not modules. What is the right action to take?

[image]

poeschlr, 2019-08-31:
The RF_WiFi library was added in https://github.com/KiCad/kicad-footprints/pull/1154

This was done as suggested by @myfreescalewebpage. This means we now need to move all wifi module from the rf_modules lib into this new one. (Not sure this is a good idea during a minor release cycle. especially as the rf_modules lib is not really that large to merit immediate splitting.)

myfreescalewebpage, 2019-09-01:
Yes I have initiated this, GPS, GSM, etc was already here.
Module with WiFi + Bluetooth should probably be kept in RF_Module... or maybe RF_WiFi_Bluetooth ?
Anyway, I agree with @poeschlr that moving in minor release is not great, we should keep like that at the moment.
Joel

gmsotavio, 2019-09-09:
I've got the point. Thanks for the clarifications. As @myfreescalewebpage, I also agree that moving at that time is not the best option. Maybe we can address this to v6.

Otavio

closed 2019-09-09 by gmsotavio

antoniovazquezblanco, 2019-09-10:
It would be nice to have an issue with the required tasks to be performed and with milestone v6.0.0 assigned so that we can keep track of this. Can any of you guys do that?

Thanks!