KiCad/kicad-footprints #1798: Rounded rectangles for exposed pads might have been a bad idea

poeschlr, 2019-09-02:
The more i think about it the more i feel that the soldermask and copper layer for non soldermask defined exposed pad pads should both be normal rectangles. (Similarly the soldermask layer for soldermask defined pads)

What is the opinion of the rest of you? If i do not get any objections then i will update the generator script within a week or so and push the updated footprints to this repo.

poeschlr, 2019-09-02:
I could also add a parameter to allow rounding for particular parts. (At least one manufacturer suggests it. See https://github.com/KiCad/kicad-footprints/pull/1691)

evanshultz, 2019-09-05:
I am sure this should be an option, as you mentioned above. All TI datasheets with thermal pads I can recall show rounded corners. As just a couple examples I could find quickly, see http://www.ti.com/lit/ds/symlink/drv8353.pdf and http://www.ti.com.cn/cn/lit/ds/symlink/drv8801.pdf.

poeschlr, 2019-09-06:
They show rounded corners but do not seem to dimension them. It kind of looks like they reuse the radius of the normal pads. So one possibility would be to offer this as a feature.