KiCad/kicad-footprints #1843: Pin 1 designator of some components is hard to spot

seppestas, 2019-09-11:
The pin 1 designator for some components is very minimal on some footprints. Especially on small-pitch IC packages and when using "lower quality" silkscreen (often as a result of ink jet systems common today) this results in a pin 1 indicator that is really hard, if at all possible to spot.

This could potentially cause issues during PCB assembly, especially when this is done by 3rd parties relying on the silkscreen.

As an example, the slightly longer silkscreen line to indicate pin 1 is only 0.3mm longer than the line on the other side for the [USON-10_2.5x1.0mm_P0.5mm footprint](https://github.com/KiCad/kicad-footprints/blob/29c6e48/Package_SON.pretty/USON-10_2.5x1.0mm_P0.5mm.kicad_mod). This looks fine when looking at a zoomed in picture on a high resolution screen, but on a real PCB this difference is really hard to spot.

I much favour the "dot style" pin 1 indicator since it's a lot easier to spot and it doesn't cause confusion like explained in #278.

IPC-7351C describes a couple of ways to mark pin 1, I assume the "slightly longer silkscreen" is based on it. However I have strong doubts it's applicable on this type of small footprint.

Since all these IPC rules are behind a paywall and thus hard to reference I think it would be interesting to clearly document proper pin 1 indication on the silkscreen in the KLC. [Right now the only things in there are](http://kicad-pcb.org/libraries/klc/F5.1/):
1. Pin-1 designator is provided on the F.SilkS layer
2. Pin-1 designator must be visible after board assembly

+ some pictures without explanation as examples. Having better, more clearly defined rules for this would make sure the pin designator is easy to spot and consistent across different types of components.

poeschlr, 2019-09-11:
Your favored dot style is in violation of IPC and will not be used by the official lib.

The footprint you show is however also in violation as the line should be removed near pin 1 instead of lengthened (as this is a no lead package, only gull wing style packages have silk elongated.)
I however doubt this will make it more visible in this case. There is simply a limit how small a part can be for silk to be a reliable helper. This is where the fab layer then comes in.

seppestas, 2019-09-11:
@pouschlr Note that the second image for KLC rule F5.1 also shows what I assume is an elongated edge near pin 1.

Keep in mind that not everyone has access to the IPC rules, and they tend to change. When they are used or enforced, IMO it's also important to mention what rule is used. I think the main problem is that the KLC does not clearly define how pin 1 should be indicated on the silk layer. Other parts of the KLC much more clearly define what should be done and what IPC rules are used.

Silkscreen is still often used as a reference, especially for low to mid-density PCBs. Having information on there that is not visible in real life makes no sense to me. Following the same argument, font sizes on the reference designator should also be tiny to the point where they are barely readable.

The "dot style" pin 1 designator can easily be removed for dense designs, just as the font size and thickness can be reduced for the reference designator or the reference designator could be removed.

poeschlr, 2019-09-11:
The general idea of the library is that we slowly move towards more and more IPC compliance. Using dot style would move in the opposite direction so it is not going to happen.

You are however correct that the rules can be documented a bit better. (It is on my list of things to do. But a day only has so many hours.)

Also re read my post! i said extended silk is used for gullwing style packages. The main screenshot clearly is of a gullwing style package (it is of a qfp.) The second part with extended silk is either a 5 pin sot gullwing package or a soic style gullwing with missing pin 5. I think it would be clearer to see if the fab layer would be shown in the screenshot.