KiCad/kicad-footprints #1862: What is the reason for Horizontal/Vertical and not Angled/Straight

evanshultz, 2019-09-16:
Blast from the past...

Does anyone recall or can find why this name change took place? It appears Rene suggested this at https://github.com/KiCad/kicad-library/issues/1622#issuecomment-334839536 (Oct 2017) and it was being enforced by https://github.com/KiCad/kicad-footprints/pull/7 (7 Nov 2017).

I seem to remember there was a discussion about which terminology worked best, and the key to the decision rested upon the word "angled" not being specific (could be 90 degrees, 45 degrees, etc.).

However, I can't find that conversation. Does anybody recall if I'm correct or can find it?

In hindsight, `Straight` and `RightAngle` might have worked and are the same number of letters as `Vertical` and `Horizontal`. I think the current paradigm is most salient and works well, but I just happen to notice that when I was searching.

poeschlr, 2019-09-17:
Angled/straight don't make sense for all things. They can also be confusing (especially if you then also get 45° parts into the mix. They are after all angled.)

Another alternative (for connectors) would have been side and top entry (better as neither would fit the 45° parts so no confusion here)

In the end it was something where multiple choices were available and we needed to take one of them. Horizontal/Vertical simply won out at the time and i would not change it to be honest. (Massive renamings always come at the cost of long term users whose own assets are now out of date. So there really must be a major benefit to it for it to make sense. I do not really see a reason to exchange the current orientation naming.)

evanshultz, 2019-09-17:
All I wanted to know was about the discussion on this. It seems we recall the same discussion and outcome. Thanks for confirming.

closed 2019-09-17 by evanshultz

poeschlr, 2019-09-18:
The thing is: If we question all of our decisions every few months then we will go crazy within a very short amount of time. (Or one of us burns out. Likely me)