KiCad/kicad-footprints #1869: Diode_Bridge_Vishay_MBLS footprint wrong pin number and pad

AndreyFursov, 2019-09-22:
pin 2 and 4 is replased. Pad rotated

poeschlr, 2019-09-22:
What symbol are you using? This footprint is paired with for example MBL104S.

AndreyFursov, 2019-09-22:
I'm find it  then paired with Device:D_Bridge_+AA-. This pin order is equal [Vishay datasheet](https://www.mouser.com/ds/2/427/mbl104s-245342.pdf)

chschlue, 2019-09-22:
The footprint doesn't match Vishay's numbering (+AA-) but matches the MBL104S symbol (+-AA).

AndreyFursov, 2019-09-22:
If footprint named as "Vishay MBLS" it mast had pin order by Vishay datasheet and symbol too. For example, ONsemi bridges have a different number order of pins, but compatible pin position.

cpresser, 2019-11-01:
I agree with @AndreyFursov and made some patches to fix this.

DanSGiesbrecht, 2019-11-02:
While I dislike how Vishay has chosen to disregard the convention of "counterclockwise from pin#1", I agree that deviating from the datasheet adds additional confusion.

This is not a breaking change, as long as the symbol and footprint library PRs are merged at the same time. @poeschlr, if you agree, I'll go ahead and merge the two PRs, and close this issue.

closed 2020-05-12 by chschlue