KiCad/kicad-footprints #1896: BatteryHolder_Keystone_3034 does not make contact

5N44P, 2019-10-06:
I've manufactured some board with that footprint, but the contact with the battery fails, probably because of the soldermask thickness. 

Should i make a PR with a larger pad, as i made in #1622?

chschlue, 2019-10-06:
With planar surface finishes, maybe even increasing it to match the cell's negative terminal diameter (and mask cut-out exceeding it) might be a good idea?

poeschlr, 2019-10-06:
What does the battery holder manufacturer suggest? After all we build footprints after the manufacturers suggestions not necessarily to fit the constrains of one particular pcb manufacturing process.

chschlue, 2019-10-06:
http://www.keyelco.com/product-pdf.cfm?p=798

[image]

Looks ridiculously tiny to me.

chschlue, 2019-10-06:
IEC 60086-3 cell dimensions:

[image]
[image]
[image]

5N44P, 2019-10-06:
I'm having the problem even with HASL, so it's definitely a big issue here. I think we should not stick with the minimum. 

@chschlue i'm not very familiar with the IEC standards. For what i see, a cell with a 20mm that goes under the commercial de nomination of cr2032 has a negative contact diameter much bigger that 8mm. And if i recall correctly, the keystone 3034 is sold for those cells. 

I think a pad as big as the inner negative contact would work for sure, with no significant risks of short circuits

chschlue, 2019-10-06:
I also think we should not stick with the minimum.
d4 in IEC 60086 is the minimum negative terminal diameter, the maximum depends on the cell's overall height:
[image]

All button cells I've ever seen appear to have a diameter close to the maximum and as it is (or should be) impossible to short it on the negative pad unless it is inserted in an angle over 45°, I don't see why the footprints shouldn't go with something in that neighborhood.

I can't publish the whole standard here but the screenshots should catch the essence.

chschlue, 2019-10-06:
On a different note: Shouldn't battery holders have courtyards large enough actually insert the battery, analogous to connectors having enough clearance for mating?

chschlue, 2019-10-06:
> I'm having the problem even with HASL, so it's definitely a big issue here.

Missed that before. So there seems to be no common manufacturing process left that could work reliably with the MF min recommendation.

5N44P, 2019-10-06:
> So there seems to be no common manufacturing process left that could work reliably with the MF min recommendation.

Maybe it's intended for a stencil that leaves paste also on the negative pad. But i didn't made any stencil, and at the moment i can't check if that pad is set for leaving the paste on the Paste layer

chschlue, 2019-10-06:
Keystone 3000 and 3034 as well as Linx HLD-012 have paste on pad 2, but I assume that's an oversight.
Linx recommends a much larger pad, BTW.
https://linxtechnologies.com/wp/wp-content/uploads/bat-hld-012-smt.pdf

5N44P, 2019-10-07:
Also for 20mm cells: 
https://www.mouser.it/datasheet/2/238/bat-hld-001-1659379.pdf
This specific part looks like should be compatible with the keystone 3034 footprint, but with a 17.8mm pad

PR #1898 (2019-10-19): Fixed coin cell holder negative contact size, removed from paste layer
Fixes #1896, see the issue for the discussion

All the changed dimensions are from linx parts documentation ([12mm][1], [20mm][2]), except the one for 24mm cells, which is an estimate based on IEC 60086, since there are no datasheets with reasonable (and not **minimum**) size specified.

Screenshots:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

[1]: https://linxtechnologies.com/wp/wp-content/uploads/bat-hld-012-smt.pdf
[2]: https://www.mouser.it/datasheet/2/238/bat-hld-001-1659379.pdf
---

All contributions to t [... cut]
closed 2019-10-19 by chschlue