KiCad/kicad-footprints #1902: ESP-WROOM-02 Pad Width too small

forkineye, 2019-10-09:
Part in question: https://github.com/KiCad/kicad-footprints/blob/master/RF_Module.pretty/ESP-WROOM-02.kicad_mod

Datasheet: https://www.espressif.com/sites/default/files/documentation/0c-esp-wroom-02_datasheet_en.pdf

Issue:
In the footprint, the pad widths are .85mm.  The recommended land pattern on page 19 of the datasheet specifies 1.5mm.

poeschlr, 2019-10-09:
The suggested land pattern is new in the datasheet. We orignally needed to rely on IPC formulas to calculate it. See: https://github.com/KiCad/kicad-footprints/pull/767#issuecomment-408362758

forkineye, 2019-10-09:
There are few other minor adjustments I'm finding as well.  Should I submitted a PR on this once I'm doing editing it?