KiCad/kicad-footprints #1904: SOIC-4_4.55x3.7mm_P2.54mm is named wrong (is a soic-6 with missing pins)

poeschlr, 2019-10-09:
Discovered in the process of working on https://github.com/KiCad/kicad-footprints/pull/1821

evanshultz, 2019-10-09:
Ummm... I'm not sure about that.

A SOIC-4 with 2.54mm lead spacing is what we have here. I find this footprint and it's name perfectly valid.

But https://www.mouser.com/datasheet/2/408/TLP185_datasheet_en_20190617-708541.pdf shows a 6-pin part. If so, the footprint should specify 1.27mm lead spacing and exclude two pins.

So my take is that the footprint you mention is totally valid, but should be generated by a different definition. And to make a footprint that properly matches the datasheet for TLP185 we need a tweak to the existing definition.

poeschlr, 2019-10-10:
A SOIC-4 would have pin numbers 1-4 not 1-6

evanshultz, 2019-10-10:
What I am trying to communicate, and apparently not doing a good job of, is saying the pitch is also wrong. If it's six pins with two removed, pitch is 1.27mm. If it's really only four pins, pitch is 2.54mm.