KiCad/kicad-footprints #1920: Connectors_Card:MicroSd_Wurth_693072010801 footprint issue

IMMRMKW, 2019-10-13:
While adding a 3D model to this KICAD footprint, I found out some small layout errors. The distance between pad 4 and the lower right pad 9 should be 6.8, and is 6.9. The footprint works, as I have tested in real life already, but it is not according to the datasheet.

gmsotavio, 2019-10-13:
I'm verifying the reported issue.

gmsotavio, 2019-10-13:
Despite the wrong courtyard clearance, I was unable to find any deviation from the datasheet.
[image]

[image]

IMMRMKW, 2019-10-14:
I could be wrong, but when I look in the datasheet, the distance between pin 4 and pin 9 is 6.8mm, and if I open the respective pins in the footprint Connectors_Card:MicroSd_Wurth_693072010801, I see X=-0.025mm and X=6.875mm, so that would add up to 6.9mm, right? Or am I missing something?

[image]

gmsotavio, 2019-10-14:
Hi, @IMMRMKW I believe you have been taking an incorrect reference. In my understanding, 6.8mm does not take into account the distance between the center of the pad 9 to the center of the pad 4, but the center of the pad 4 to the edge of the card connector.

IMMRMKW, 2019-10-14:
Ah! I am sorry, I think you are correct. I was also using this as a reference for my 3D model so that is probably why it was still misaligned.

gmsotavio, 2019-10-14:
At least it emerged from this discussion that the courtyard clearance is wrong. It can now be fixed. Thanks.