KiCad/kicad-footprints #1942: SMD pads for castellated edge modules

gmsotavio, 2019-10-27:
Hi people,

I have a doubt about SMD pads for castellated edge modules. Most manufacturers of such modules often do not provide a recommended land-pattern for their modules. We have some open pull requests and footprints in current kicad library that fall into this category. My question is how to review them. Is there something about that in IPC-7351 or another related standard?

poeschlr, 2019-10-27:
IPC-7351B has suggestions for castellated pads on page 17.

Nominal density: heel 0.55, toe 0.15 side 0.05
Strangely heel for castellated is defined as pointing outwards.

gmsotavio, 2019-10-27:
I believe you are talking about **Table 3-9 Leadless Chip Carrier with Castellated Terminations (unit: mm)**.

Thanks.

poeschlr, 2019-10-27:
@gmsotavio did you delete your comment again or is there a problem with github? (The dashboard shows me a comment but there is none here)

gmsotavio, 2019-10-27:
I made a comment asking about the information you said was available on page 17, but I remembered that I don't have access to IPC-7351B here, just the old IPC-7351A. So I deleted it because it didn't make sense. Anyway, thanks for the info.