KiCad/kicad-footprints #1955: RJ45 connectors all have different styles

cp-aquila, 2019-11-06:
This came up while looking at: https://github.com/KiCad/kicad-footprints/pull/1954

Most of the connectors do not have a similar graphic style.
- [ ] Pin1 Silk-Marker is quite consistent, most have the triangle. Some however have a 'remove silk-line next to pin1' style. Some have a short duplicate silk-line.
- [ ] Pin1 Fab-Marker: there are two major styles. clipped-rectangle and arrow. I suggest to use arrows everywhere.
- [ ] Courtyard clearance. We have 0.25mm, 0.5mm and 2.5mm!! KLC is pretty clear here, they should all be 0.5mm

Pin-Numbering is also pretty wild:
- [ ] Some have LEDs with prefixed pin-numbers
- [ ] Shield handling is different.

Some do not have a proper description (Just a datasheet link)

gmsotavio, 2019-11-06:
> Pin1 Silk-Marker is quite consistent, most have the triangle. Some, however, have a 'remove silk-line next to pin1' style. Some have a short duplicate silk-line.

I noticed the same thing. But KLC says nothing about SIlk style. I imagine it would be interesting to propose a change in KLC to standardize the style.

> Pin1 Fab-Marker: there are two major styles. clipped-rectangle and arrow. I suggest to use arrows everywhere.

Usually, a bevel should be used when the pin-1 is near the edge of the footprint outline and a chevron when the pin-1is located among other pins or in the middle of the fab outline. But again, KLC says nothing.

> Courtyard clearance. We have 0.25mm, 0.5mm and 2.5mm!! KLC is pretty clear here, they should all be 0.5mm

In this regard, there is no reason to doubt, KLC is pretty clear.
-> 2.5mm. I believe you are talking about `RJ45_Plug_Metz_AJP92A8813`. Moreover, the datasheet link is missing.

[image]
[image]

> Some do not have a proper description (Just a datasheet link)

I noticed hundreds of footprints with a missing description in the current KiCad library, but usually in old footprints.

cp-aquila, 2019-11-06:
Up-To-Date Metz-Connect Datasheet: https://www.metz-connect.com/de/system/files/productfiles/Datenblatt_AJP92A8813.PDF
This connector is not a jack, but a board mounted plug. So having a big courtyard for the required mating connector is a good choice. As stated by rene in #878 its is unfortunate that no explicit counterpart is given, so it is not clear how big the courtyard should be.
We could improve it by adding some text on Cmt.User and/or more Fab lines.

I would like to discuss the styles for RJ?? connectors in this issue and find a consensus. 
Then we can implement it, and perhaps write it down somewhere. I don't think we should add specialized info like this into KLC itself. Its already pretty big. Is a KLC appendix, one for each class of parts a good idea? This would also be a solution for some of the other 'unwritten rules'.

- I don't have access to IPC documents (my google-foo is failing), perhaps someone with insight can have a look if there is a recommended style.
- F.Fab I like the chevron for pin1. Reasoning: For most FPs the bevel is to far from the actual Pin1.
- F.Silk pin1: You are right, KLC does not have a distinct style. [image]
  - I like the chevron/triangle because its is easily visible. But its large, so just having one additional line seems like a good choice

gmsotavio, 2019-11-06:
> I don't think we should add specialized info like this into KLC itself. Its already pretty big. Is a KLC appendix, one for each class of parts a good idea?

On second thought, you're right. Perhaps include an additional example in the image below suggesting a preferred style for silk in such connectors to avoid mixed styles.
[image]

cp-aquila, 2019-11-07:
This .pdf was shared in another issue, and it also seems usefull here: https://ocipcdc.org/archive/What_is_New_in_IPC-7351C_03_11_2015.pdf#page=18

Both screenshots do not explicitly apply to THT parts, but bottom-only seems to be close enough.

For Silkscreen:
[image]

Fab-Layer:
[image]

@evanshultz Afaik you have access to the whole IPC-Document. Can you shed some light on this?

chschlue, 2019-11-07:
#1953