KiCad/kicad-footprints #1962: DFN-10-1EP_3x3mm_P0.5mm_EP1.65x2.38mm silkscreen has wrong pin 1 indication

mansr, 2019-11-12:
The pin 1 indication on the silkscreen of DFN-10-1EP_3x3mm_P0.5mm_EP1.65x2.38mm is on the wrong side. See screenshot below:
[image]

Compare to the EP1.55x2.48mm variant:
[image]

JacobEFO, 2020-03-04:
Hi @mansr,
I have stumbled upon the exact same issue as you: https://github.com/pointhi/kicad-footprint-generator/issues/278

However for non-gullwing packages it is standard to have a missing line at the pin 1 designator, just as the upper footprint you illustrate. However, that would in turn mean, that your second footprint is not shown according to KLC.

However, because of the amount of work, fixing the non-gullwing package footprints is just on-going work and should eventually be fixed.

Disclaimer: Not a librarian, so if you prefer more elaboration. I'm confident the librarians will chime in.

poeschlr, 2020-03-04:
@JacobEFO is correct. The upper footprint is the one that complies with IPC which is what the KLC suggests to use. See screenshot in https://kicad-pcb.org/libraries/klc/F5.1/ (The bottom right one is of a DFN footprint)

closed 2020-03-04 by poeschlr

mansr, 2020-03-04:
Well, that's not confusing at all. Shouldn't all DFN footprints at least use the same convention? Otherwise it all becomes rather pointless.

JacobEFO, 2020-03-04:
> Well, that's not confusing at all. Shouldn't all DFN footprints at least use the same convention? Otherwise it all becomes rather pointless.

You're absolutely right, they should. 
But unfortunately such extensive work takes time, and as you may be able to see, maintaining such large repositories, that we all gladly utilize, takes quite a while. 
I reckon, if this is not set for a specific milestone, that any help in this regard will be greatly appreciated.

chschlue, 2020-03-05:
BTW:
Both FPs shown above don't meet the current standard (one has a roundrect EP, the other has rectangular pads.)
The first has to be regenerated, the second added to the generator.

poeschlr, 2020-03-05:
@chschlue we have done this switch to rectangular center pad after the last release so only users who manually download the libs have them.

chschlue, 2020-03-05:
@poeschlr Ah, yes. My mistake.

DFN-10-1EP_3x3mm_P0.5mm_EP1.65x2.38mm is already fixed in master.
However, DFN-10-1EP_3x3mm_P0.5mm_EP1.55x2.48mm is hand drawn (in case anybody wants to add it to the generator size definitions at https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/Packages/Package_NoLead__DFN_QFN_LGA_SON/size_definitions/dfn.yaml)

poeschlr, 2020-03-05:
Nearly all DFNs are still not in the generator. I suspect it will take at least a man month (160 man-hours) of work to go through all of them (most of the time will be used to find datasheets and check if the symbols that use the current footprints are assigned the correct ones.)