KiCad/kicad-footprints #1983: GitLab Footprint Migration

sethhillbrand, 2019-11-27:
Hi Librarians-

As part of the Great KiCad GitLab migration of 2019, I am testing the movement of open issues and PRs from GitHub to GitLab.

Here's what we know so far:

- Open Pull Requests will transfer to GitLab without moving the source of the PR.  That means pull requests will become GitLab Merge Requests that are referenced to the user's GitHub repository
- Open Issues will transfer to GitLab issues
- Closed/Resolved PRs and Issues will also transfer

In order to facilitate the transfer, it will be helpful if you can do two things:
- Open a GitLab account (these can use your GitHub OAuth)
- Set your GitHub email to public

The last point is critical because the migration script will only link users whose e-mail addresses match.  If the script can't see your GitHub e-mail address, the issue will be tied to the KiCad-Bot that performs the migration.

The timeline for the actual migration will be determined by the librarian group but it would greatly assist my testing for bugs if you could do the two steps above as soon as possible.

Thank you-
Seth

evanshultz, 2019-11-27:
@poeschlr @pointhi 
I don't see a better place to do this, so hopefully this isn't a bother to add this tangent...

Would the GitLab migration be an opportune time to revisit making https://github.com/pointhi/kicad-footprint-generator a first-party repo? Or at least moving that repo to GitLab as well? Thomas, if you're willing to contribute your work that would be great as it's critical infrastructure to the footprint libraries.

pointhi, 2019-11-28:
@evanshultz I would be okay with this. While doing so, I would like to split the repo into two parts:
 one is the library itself, and the other one for all scrips which generate the footprints. While doing so, I will try to create a release on pip, so people only need to clone the scripts folder.

chschlue, 2020-08-31:
Obsolete, closing.

closed 2020-08-31 by chschlue