KiCad/kicad-footprints #2000: R_2512 new footprint

jneiva08, 2019-12-10:
Hi,

I found another type of footprint for R_2512 ([datasheet](https://www.te.com/commerce/DocumentDelivery/DDEController?Action=showdoc&DocId=Data+Sheet%7F9-1773463-7%7FE%7Fpdf%7FEnglish%7FENG_DS_9-1773463-7_E.pdf%7F2-2176230-0)), the issue is the pads are more wide than the standard.
What is the best way to resolve? PR for  footprint hand made or exists footprint script for the resistors?

How can name it? R_2512_6332Metric_TE_CGS

Best regards

evanshultz, 2019-12-12:
While any given vendor may come up with their own footprint, the library is currently using IPC standards to generate footprints from the package dimensions. The vendor footprint isn't considered.

The size definition for 2512 at https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/SMD_chip_package_rlc-etc/size_definitions/size_default_chip_devices.yaml includes the length and width, including the tolerance, of the datasheet you sent.

So I think you'll be fine using the existing footprints.

While there are reasons to add manufacturer-specific footprints, such as if this was a 4-pin 2512 sense resistor with Kelvin terminals, in this case I'd push back against it.

poeschlr, 2019-12-12:
I agree with evan here. This really is not something that we would include in the official distribution. I intent on transfering the scripts i wrote as a footprint wizard such that users can create their specialized footprints easily against their manufacturing tolerances and the tolerances of the part they have decided to use.

closed 2019-12-12 by poeschlr