KiCad/kicad-footprints #2001: Capture keepouts as polygons?

evanshultz, 2019-12-12:
Currently KLC says keepout areas should be noted with lines https://kicad-pcb.org/libraries/klc/F4.5/.

I'm sure the devs are smart enough to implement keepouts with polygons since that's what they'll be. The line method currently in KLC doesn't define a closed shape. And polygons may not have been available in KiCad when this KLC clause was written.

I propose KLC is updated to define keepouts as polygons. Thoughts?

You can see an example of what this would look like at https://github.com/KiCad/kicad-footprints/pull/1885#issuecomment-552001858. It also makes the text for the keepout type appear much more obvious (naturally this depends on the color settings).

poeschlr, 2019-12-12:
Is there a way to have polygons as non filled? If not then i vote for continuing the use of lines.

evanshultz, 2019-12-12:
Not that I can see now. But https://gitlab.com/kicad/code/kicad/issues/3662 is asking for this capability.