KiCad/kicad-footprints #2013: Best way to add more mounting holes?

cnieves1, 2019-12-19:
Hi,
I wanted to add some more mounting holes to the library, for M1.4 and M1.6mm.

I found there is a script to generate them, at:
https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts/Mounting_Hardware

and also an example yaml file at:
https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/Mounting_Holes/mounting_hole_long.yaml

However, this is not the yaml file used to generate all the mounting holes in official Kicad library.

Could you tell me where is this yaml file or the script used to generate the official mounting holes?

Thanks,

myfreescalewebpage, 2019-12-19:
I'm also interested in this subject because I need more mounting holes too. probably we can generate mounting holes every 0.1mm from 1 to 10mm right now.

@poeschlr and @evanshultz can you help on the question of @cnieves1 ? How the mounting holes existing in the library have been generated ? Need to fix existing one with the generator ?

Thanks!

poeschlr, 2020-01-31:
I think that script was not used to generate any of the current mounting hole footprints. I do not know if it can generate footprints of the current style used for the included mounting holes. If it does then i see no reason why it should not be used to add future mounting holes and possibly even to replace the current footprints to ease maintainability.

If you add something to it then please base it on my cleanup branch as i moved the yaml file into the same directory as the script https://github.com/pointhi/kicad-footprint-generator/pull/487

myfreescalewebpage, 2020-09-03:
@cnieves1 have you work on some scripting method for the Mounting Holes ? I clearly need more footprints, particularly with some paste options (soldering of spacers like https://www.we-online.de/katalog/datasheet/9774060243R.pdf) so I'm thinking to doing this quickly

chschlue, 2020-09-04:
@myfreescalewebpage I believe what you're looking for is in `Mounting_Wuerth`

cnieves1, 2020-09-04:
@myfreescalewebpage I'm sorry but I have no update about this...

myfreescalewebpage, 2020-09-04:
@chschlue thanks to point this, really happy to found them!