KiCad/kicad-footprints #2016: Raspberry Pi footprints

jstjst, 2019-12-22:
I started to create the footprints for the RPi symbols. Symbol Issue: https://github.com/KiCad/kicad-symbols/issues/1772
There was a previous discussion here: #1064 

I first looked over the existing RPi Zero footprint (PR #1066 ).
I added the micro SD card outline to the footprint, the rest of it is fine.

I added the Run header, but the outline graphics overlap the GPIO header.
Should I extend the outline graphics from the GPIO header and "merge" them together?

[image]

jstjst, 2019-12-22:
List of all symbol PR:
- [ ] https://github.com/KiCad/kicad-symbols/pull/1820
- [ ] https://github.com/KiCad/kicad-symbols/pull/1850
- [ ] https://github.com/KiCad/kicad-symbols/pull/1851
- [ ] https://github.com/KiCad/kicad-symbols/pull/1852
- [ ] https://github.com/KiCad/kicad-symbols/pull/1853
- [ ] https://github.com/KiCad/kicad-symbols/pull/2397

jstjst, 2019-12-25:
@myfreescalewebpage Can you take a look at the problem shown in the picture above?

myfreescalewebpage, 2019-12-26:
@jstjst the silkscreen lines should be visible so I suggest you should remove the part of the lines under the connector, just between pins 37/42/39/41 and merge the lines.