KiCad/kicad-footprints #2078: Should (solder) jumper footpirnts define a mask cutout between their pads

poeschlr, 2020-01-31:
See https://forum.kicad.info/t/should-jumper-pads-have-solder-mask-between-them/20998

The argument for adding an additional mask only pad is that it would be easier to cut the trace or to solder it shut (depending on type). I do not know if there would be downsides.

poeschlr, 2020-01-31:
From the forum: wave soldering and reflow soldering might be troublesome without mask in between. So an option would be to create a separate set of footprints for hand soldering that do not have mask between the pads.

evanshultz, 2020-01-31:
Separate footprints, both with and without mask, is a good idea.

I have used an open jumper with no mask, on the bottom of a board, that shorts during wave solder to make a desired connection. But if service is needed, it's easy to wick away the solder. Another example of why no-mask footprints can be helpful. In this case it was a _solder_ jumper with the pads and area between the pads exposed, but non-solder jumpers could have mask removed only between the pads.