KiCad/kicad-footprints #2082: MSOP-8_3x3mm_P0.65mm has too narrow clearance

essele, 2020-02-07:
This MSOP-8 footprint seems to have a pin to pin clearance of approximately 5.8mil, which will cause it to fail DRC checks with a 6mil clearance set.

All of the other MSOP-8's (the EP variants) all have a clearance approaching 10mil which seems a lot more sensible.

(Pins are 0.5mm on this one, and 0.4mm on the EP variants -- most of the datasheets I've looked at suggest between 0.35 and 0.45 ... so 0.4 seems like a good choice.)

andrewteasdale, 2020-02-11:
I've just found the same thing on exactly this footprint and settled on 0.45mm pad width, which matches some of the 0.65mm pitch TSSOP parts. But generally there is not a lot of consistency across these footprints, with some having rounded corners per latest IPC and other not etc. I'd take on the challenge of harmonising this but some of the variation may be for a reason and settling on a defined standard for this will require a library manager to confirm an approved schema I suspect.

evanshultz, 2020-02-11:
We are going with nominal-density IPC land patterns, but not all gullwing packages have been moved to the generator, as mentioned above. Contributions are welcome if you'd like to help with that effort.

andrewteasdale, 2020-02-11:
Thanks @evanshultz, I've put adding some of the missing footprints to the generator onto my task list.

chschlue, 2020-02-11:
FYI:
This particular FP has recently been added to the generator but is generated without `force_small_pitch_ipc_definition` set, contrary to most other 0.65mm pitch MSOPs.
That's why it has 100µm wider leads.

andrewteasdale, 2020-02-12:
Thanks @chschlue - and apologies in advance for the long response! 

To make this more complicated, my understanding is that `force_small_pitch_ipc_definition` should only be set for footprints <=0.625 mm, at which point the IPC-7351 side goal goes from 0.03 to -0.02 mm. If that assumption is true then a number of the existing 0.65 mm pitch script-generated parts are incorrect. 

I suspect the thing that concerns people is that a 0.5 mm pad width with only 0.15 mm clearance feels and looks a bit tight. I stuck the JEDEC part spec into an IPC calculator and came up with a 0.48 mm pad width, which is indeed 0.5 mm when rounded to 0.05 mm, which is what IPC-7351B seems to recommend. Adjusting the placement tolerance brings the pad width down to the more reasonable 0.45 mm, but that's a bit of a fudge, and in reality more like 0.42 mm makes sense [but does not fit with the 0.05 mm rounding].

Incidentally, the pads generated using the default heel/toe for a nominal environment lead to a much larger footprint than the previous library version. This is on balance a good thing as the existing footprint wasn't ideal. However, there are flaws in the IPC-7351B approach [namely a one-size-fits-all pad size] which I understand is being resolved in IPC-7351C, with differing heel/toe dimensions for different lead pitchs as follows [source https://ocipcdc.org/archive/What_is_New_in_IPC-7351C_03_11_2015.pdf]:

[image]

A good source here is the PCB Library Expert approach, which is a hybrid of what's coming in IPC-7351C and common-sense extrapolations from the IPC-J-STD-001 solder joint goal acceptability guidelines. This makes use of differing heel/toe/side values for various pitch parts on a sliding scale, rather than the big/small approach used in the current IPC-7351B approach [and thus the generator scripts]:

[image]

Following some experimentation, in the absence of an agreed approach these are the values I'd probably implement in my own libraries as the footprints look both highly solder-able and sufficiently dense for a realistic nominal environment. 

I'd be happy to get stuck in and update some of the remaining non-script-generated to 'new' IPC-style footprints as they need improving and standardising, but I suspect not everyone [myself included!] would be happy with the default 0.5 mm pad size, and hacking the scripts to generate 0.42 mm or 0.45 mm pad widths feels wrong.

Of course by far the simplest workable approach is for everyone to agree that [as an exception to the IPC spec] we move the 'small pitch' boundary from 0.625 mm to 0.65 mm and use `force_small_pitch_ipc_definition` for all 0.65 mm pitch parts. This would generate a perfectly reasonable [if narrower than IPC norms] 0.4 mm pad width and match the balance of 0.65 mm pitch parts already in the library. @evanshultz would you be happy with that [in which case I will get started]?

andrewteasdale, 2020-02-12:
UPDATE: I've just seen the excellent work @essele has done in #2090 with all the TSSOP variants based on the JEDEC standard, which is exactly what I was hoping to implement for some of the other SOP families [unless this work is already planned by someone else?]. 

This PR has worked out perfectly for 0.65 mm pitch variants because the lead width on TSSOP parts is narrower than for e.g. MSOP parts, and thus `force_small_pitch_ipc_definition` is not required to product an acceptable pad width. Fantastic job!

chschlue, 2020-02-14:
IMO, fine pitch goals are fine for MO-187 AA MSOPs,
Should this be set manually or should the script be changed to use these for all <=0.65mm pitch FPs?
MS-026 0.65mm TQFPs have equally wide leads, for example.

To answer your other question: AFAIK, #2090 is the only SOP rework at this time.

evanshultz, 2020-02-14:
A few things pop to mind...
- We currently have an IPC 7351B generator (with round rect pad shape from early 7351C info) but that standard is supported by external files so adding other standards should be possible without too much effort.
- 7351B is now a decade old and 7351C seems to be stuck in purgatory so it probably is worthwhile to consider if some tweaks are reasonable for this library to keep up-to-date with equipment capabilities and user desires of 2020.
- However, I believe it will be overwhelming for no real benefit if many more variants are offered which are all very similar. So I would prefer to limit things to just IPC 7351 footprints, with the full three density levels where helpful to users. Making small tweaks like the above is no problem for me as long as we don't add footprints which aren't useful for users or near-duplicates that add clutter.
The `force_small_pitch_ipc_definition` property isn't needed because the finer-pitch IPC rules are used if the lead pitch is <=0.625mm.
- Moving the break on gullwing parts from 0.625mm pitch up to 0.65mm pitch seems fine with me, but I'd rather do it using the property on the package entry so that we're conforming to the IPC rules as they are now and we also have a place to capture this intentional deviation.

My brain may not be working. What do y'all think?

poeschlr, 2020-02-15:
The `force_small_pitch_ipc_definition` is kind of a hack that allows to overwrite the automatic definition for pitches that are very close to the cut-off point. I simply noticed that 0.65mm footprints often result in way too little clearance and would be better off with the smaller definition.

PR #2102 (2020-02-22): Generate MSOP-8_3x3mm_P0.65mm using fine pitch definitions
Fixes #2082

Script https://github.com/pointhi/kicad-footprint-generator/pull/502

[image]

---

All contributions to the kicad library must follow the [KiCad library convention](http://kicad-pcb.org/libraries/klc/)

Thanks for creating a pull request to contribute to the KiCad libraries! To speed up integration of your PR, please check the following items:

- [ ] Provide a URL to a datasheet for the footprint(s) you are contributing
- [x] An example screenshot image is very helpful 
- [ ] If there are matching symbol or 3D model pull  [... cut]
commit daa60edd (2020-02-22): Generate MSOP-8_3x3mm_P0.65mm using fine pitch definitions (#2102)

Fixes #2082
closed 2020-02-22 by poeschlr