KiCad/kicad-footprints #2146: ZH line missing from Connectors_JST library

Teslafly, 2020-03-16:
you have dome a great job with most of the connector libraries, but I find that a major omission is the ZH line (1.5mm through hole). Which is the through hole version of the ZE line.
I would make and commit these myself, but from what I've read so far, the generators and committing to this library has a relatively high learning curve.

evanshultz, 2020-03-17:
Thank you. I hope you are finding the libraries good otherwise.

If one knows Python, the scripts are simple. If one doesn't, it's a great exercise to learn. Should ZH be just like ZE except for the pins, it's likely a relatively simple change to the existing script.

We don't take requests since that would really overwhelm the librarians and contributors, so I'm going to close this. It will stay around if someone else finds it later and wants to develop the script. Or if you want to give it a try, the existing ZE scripts are at https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts/Connector/Connector_JST.

closed 2020-03-17 by evanshultz

Teslafly, 2020-03-23:
took a shot at modifying the generator now that i actually found it. based on the ph generator.
https://github.com/Teslafly/kicad-footprint-generator/blob/master/scripts/Connector/Connector_JST/conn_jst_zh_tht_top.py
the silkscreen is currently not fixed, but pins are ok.

How do you use kicad-footprint-generator to generate the whole kicad-footprints library?

also, where is the generator for the JST_ZE_SM14B-ZESS- series connectors? all I can find are the through hole variants?