KiCad/kicad-footprints #2149: Battery contacts - which library?

mibe, 2020-03-16:
Hey guys,

I have two new footprints ready, they are the 9 volt battery contacts Keystone [269](https://www.keyelco.com/product.cfm/product_id/1077) & [261](https://www.keyelco.com/product.cfm/product_id/1073). My question now is, which is the most fitting library for them?

Of course the ```Battery``` library with the ```BatteryHolder``` footprints first comes to mind. But on a second thought, these parts are _contacts_, not _holders_. They have (beside the actual contact) no mechanical support for holding a battery. You'd also need both to connect a battery to and you'd as a designer also need to get the distance of both parts on your PCB right.
My next idea was to maybe put them in the generic ```Connector``` library? They are mating contacts after all, so a designer could use them for whatever they like, as a fancy board-to-board interconnect for example.

What's you opinion on this?

evanshultz, 2020-03-17:
They're specifically for batteries, so put them in the `Battery` repo. Instead of `BatteryHolder_` as a prefix, which I still think might be reasonable, you could use `Contact_` or some other term you think might fit better.

mibe, 2020-03-21:
Alright, the ```Battery``` library it is. I'd prefer the prefix named ```BatteryContact_``` actually.

cp-aquila, 2020-04-09:
I assume the contacts always get used in pairs? I see some benefit in having them as seperate symbols+footprints as it makes the BOM creation easier.

**Question:** How does the designer know how much distance needs to be between the contacts?
**Related:** How does the designer know how big the battery will actually be? There is no keepout or courtyard associated with those contacts.

The first solution that comes to mind is to add some kind of marker in F.Fab for the 2nd contact into each footprint (maker example here https://github.com/KiCad/kicad-footprints/pull/1394). Perhaps also add the outline of the battery into each footprint. On the board they should perfectly overlap.
That leaves the courtyard. No clue on how to fix that :/

mibe, 2020-04-12:
> **Question:** How does the designer know how much distance needs to be between the contacts?

The designer would have to look up the distance in the specification / battery datasheet. It is half an inch. Maybe it'd good to have the distance in the footprint description?

> **Related:** How does the designer know how big the battery will actually be? There is no keepout or courtyard associated with those contacts.
>
> The first solution that comes to mind is to add some [...]

Indeed that's a problem. How about having an X drawn on Fab or Cmts in both footprints, which have to overlap to match the correct distance?
The courtyard could be made half the battery, so that the lines of both footprints have to overlap to match, like in the screenshot below. But how does the DRC react to having a CrtYrt line on another?
On the other hand there's so much space between the contacts and in the Z direction to the battery itself, you could easily fit standard 1206 components between the contacts. My quick measurements yielded at least 3 mm clearance in Z. So, I dunno, is a clearance including the battery itself needed? Depends on the design of the PCB I guess...

[image]