KiCad/kicad-footprints #2176: Mini PCI Express

jneiva08, 2020-03-26:
Hi,

Have made full connector size card for MM60-52B1-E1-R650 ([link](https://www.jae.com/en/connectors/series/detail/product/id=68907&type_code=T1110)
I like to know the right place to put it.

Thanks

cp-aquila, 2020-04-09:
Hi,

that could fit into `Connector_PCBEdge`. For example there is also `BUS_PCIexpress` in there.
However, if you made the full card, including edge-cuts that is not a footprint anymore but a template.
https://github.com/KiCad/kicad-templates

jneiva08, 2020-04-09:
hi, it is only for the sockts, so will be better to be in Connector_PCBEdge.

[image]

cp-aquila, 2020-04-09:
That looks nice!
As a designer I would like to put low-profile devices between the two parts.
I wonder how to properly handle courtyard for this. A split-courtyard combined with a z-keepout seems like a good idea. But I amnot sure if kicad supports a split courtyard.

Another point, if this is one footprint that corresponds to one symbol there will be only one entry on the BOM. But for ordering parts and pick-n-place two lines in the BOM are mandatory.

evanshultz, 2020-04-10:
I think https://github.com/KiCad/kicad-footprints/pull/1124 might work. Please check for compatibility and let me know. That was intended to be a generic footprint that would work for all connectors compliant with the mini PCIe spec.

jneiva08, 2020-04-10:
I have made more specific, because height of connector, have included the latch (full size), specific latch from same seller

evanshultz, 2020-04-11:
Ahh. Gotcha. Then I guess use the same library as the existing generic footprint?

Also, since the link you sent above is just the connector can you please make the retention mechanism a separate footprint? They are unique orderable parts and separate items on the BOM. I would be happy to see something indicating their relative placement on the board, though, so it's easy to place one of each footprint in proper relative to each other.

HackInvent, 2020-06-05:
I saw a similar thread here: https://github.com/KiCad/kicad-footprints/pull/1404