KiCad/kicad-footprints #2192: ST Morpho Connector mounting hole location error

jhalmen, 2020-04-02:
the locations of the mounting holes of the footprint
https://github.com/kicad/kicad-footprints/Module.pretty/ST_Morpho_Connector_144_STLink_MountingHoles.kicad_mod
are wrong. 
as also reported in #2045 we used those locations to 3d print a mount, and the holes don't line up.
the [official mechanical drawing](https://www.st.com/content/ccc/resource/technical/document/user_manual/group0/26/49/90/2e/33/0d/4a/da/DM00244518/files/DM00244518.pdf/jcr:content/translations/en.DM00244518.pdf) page 14 lacks a lot of information about the actual hole locations. does anybody have more info on these?

der-ule, 2020-07-20:
The manufacturing files for the Nucleo-144 boards are available here:

https://www.st.com/resource/en/board_manufacturing_specification/mb1312_manufacturing.zip

you should be able to extract the necessary information from them.

jhalmen, 2020-07-21:
awesome, thanks!

weirdly enough i have to say that the mounting holes of the footprint line up perfectly with the provided gerber files...
it must have been an error on my part then, even if i can't see what might have gone wrong back then with the print.

since the files agree with the footprint i'll close this. sorry for the noise

closed 2020-07-21 by jhalmen