KiCad/kicad-footprints #2231: CL-2025 Footprint

ChristianHirsch, 2020-04-20:
I'd like to add the [CL-2025] (https://www.torexsemi.com/file/CL-2025-02/CL-2025-02-pkg.pdf) footprint, e.g., for a XCL101 DC/DC convert. But to which library fits this footprint to?

cp-aquila, 2020-04-30:
How about `Package_LGA`

ChristianHirsch, 2020-04-30:
Thanks for your answer @cp-aquila, but what about `Converter_DCDC`?

cp-aquila, 2020-04-30:
I took a closer look:
- The `XCL101` is a DCDC converter. I don't expect any other device (outside that series) to fit on the same footprint.
- `Converter_DCDC` has mostly isolated DCDC converters. But there are also some non-isolated variants in there. It is mixed already and would allow a new non-isolated part.
- I could not find a single item in `Converter_DCDC` that is as small as the `XCL101`
- The `XCL101` is basically a LGA part
- `Package_LGA` does have the `Texas_SIL0010A_MicroSiP-10-1EP_3.8x3mm_P0.6mm_EP0.7x2.9mm_ThermalVias` which is similar to the `XCL101`

Unfortunately non of the above findings provide a really strong argument for either one of the library.

chschlue, 2020-04-30:
There was a similar discussion at https://github.com/KiCad/kicad-footprints/pull/2083.
The ultimate reasoning was basically to go by package type if there's a distinct name for it and a possibility that it's going to be used with other parts.
This case here isn't quite the same, though.