KiCad/kicad-footprints #2277: _ThermalVias packages don't connect the EP to a zone that has thermal-relief pad connections

BryanKadzban, 2020-05-11:
So the EP pad's thermal-relief style is "from parent footprint", and the footprint's style is "use zone setting".  When the zone is set to thermal-relief, the EP pad *should* have four connections out to the zone ... but the thermal vias are too close to where those connections would be, and the vias' own style disables them.

In particular, the three packages I see this on are all in Package_DFN_QFN:

QFN-56-1EP_7x7mm_P0.4mm_EP5.6x5.6mm_ThermalVias
QFN-48-1EP_7x7mm_P0.5mm_EP5.45x5.45mm_ThermalVias
Micrel_MLF-8-1EP_2x2mm_P0.5mm_EP0.8x1.3mm_ThermalVias

If I change the connection style of the four thermal vias near the middle of the edges of the EP pad (...or both vias for the MLF package) and change them all to "solid" connection style, then the zone connects to the pad and all vias, the way I would have expected.

Is there some way to make this work other than overriding the style on these four vias?  If not, should those four be set to "solid" in the footprint by default, instead of "from parent footprint"?

chschlue, 2020-05-11:
Zones really do interact strangely with thermal vias, but I think the underlying problem is that thermal vias are actually through-hole pads, not vias.

[image]

The upper zone in the screenshot above is set to solid while the lower one is set to "reliefs for PTH only".
Kicad connects those "vias" with reliefs, which isn't usually what you want (it defeats the purpose of the thermal vias, after all.

In case you _really_ want reliefs, actual vias instead of pads would solve that issue, too.

@KiCad/librarians can anybody confirm I'm not talking complete nonsense?

BryanKadzban, 2020-05-11:
They definitely are through-hole pads, yes.  (There's no other way to put a via into a footprint that I've seen.)  I suspect that having the reliefs present would be better from a soldering point of view, but that's pure speculation; this is the first time I've tried to make a board that needs the thermal dissipation so I don't know firsthand.  (I've done exposed pads before, but not with vias.)  In general, thermal reliefs don't affect current transmission, but they do help when trying to solder to the exposed copper.

OTOH, the setting doesn't matter for any of the through-hole pads -- except the ones on the outside -- because the single large EP surface-mount pad is always solid.  (On both top and bottom.)  It doesn't act like a zone at all.  So soldering will be harder regardless; this *probably* means that heat dissipation will be better.  (Assuming the ground plane gets connected.  :) )

I have no opinions about 3- or 4-layer boards, or the connection between the _ThermalVia pads and the internal layers; I've never used the third or fourth copper layer.

I can't exactly tell what's going on with the screenshot though; there seem to be at least four layers there, and I'm not familiar with the coloring.  :-/  Maybe these two screenshots make it more obvious what I mean.  Both are screenshots of a separated Edge.Cuts rectangle with a copy of a couple footprints and a zone on the B.Cu layer on the GND net.  All of the pad-49 pads of the QFN package are manually tied to GND, as well as pad 2 of the C54 footprint.  Pad 1 of C54 and pad 31 of the QFN package are tied to a +28V_A net that I'm using on the rest of the board.

On the first screenshot (from-parent-footprint.png), the style of all the through-hole pads is set to "from parent footprint".  You can see an airwire between C54 pad 2 and the QFN package's pad 49.

On the second screenshot (solid.png), I've changed the setting for the rightmost middle through-hole pad (which is selected) to be "solid".  You can see the thermal-relief connection to the big pad 49 on the B.Cu layer, and the airwire is gone.

[image]

[image]

Is that what you meant?

BryanKadzban, 2020-05-12:
*Ding*.  Oh.  Now I realize what the screenshot was getting at -- the bottom layer was **reliefs for PTH only**, not "thermal reliefs" (i.e. for all connections) which is what I assumed when trying to interpret it.  So that pad does have a full connection to the B.Cu zone, except where the through-hole pads cut it down to thermal-relief.

That's actually inverted from what I'm looking for: I'm looking for the through-hole pads to connect all the way around, while the large rectangular surface pad has thermal reliefs ... on the F.Cu layer.

But as I was writing that last sentence, I *also* just realized that the B.Cu layer really has no reason for reliefs, since nothing will be soldering to that side of the PCB at all.  So another way to solve this might be to set the big square pad (on the back layer) to "solid", instead of doing that to the mid-side PTH pads.

chschlue, 2020-05-12:
I think we're on the same page, but I'll still try to clarify:

1. Use case: Solid connection to thermal pad for maximum heat dissipation
[image]
(vias in pad, zone set to "solid")

2. Use case: Relief connection for better (hand) solderability
[image]
(vias in pad, zone set to either "relief" or "PTH-only relief")

I'm arguing that the _proper_ solution is to allow actual vias in addition to PTHs in footprints.

I haven't been able to find a wishlist item regarding such an FP format extension on https://gitlab.com/kicad/code/kicad/-/issues, which makes me wonder if my proposal is (for a reason I failed to take into account) so stupid that nobody else dared to bring it up yet.

evanshultz, 2020-05-12:
It is there. See item 3 at https://gitlab.com/kicad/code/kicad/-/issues/4067.

chschlue, 2020-05-12:
Sry, yes, I found that one. But it doesn't seem to explain why a thermal via has to be a pad instead of, well, a via.
There must be a compelling reason, but I still can't figure it out.

chschlue, 2020-05-12:
Oops, sry again. Item 3... must have missed it somehow.

BryanKadzban, 2020-05-13:
That sounds like a great way to improve the situation with more work on the footprint format and editor code, yes.

Is there a way to modify the footprints to work better in the meantime?  I assume the big rectangular pad on the bottom is supposed to be able to connect to zones in some way or another.  :)

...On the other hand, both a footprint library change and a code change will require a release before people can use them, so yeah, maybe either way works.

chschlue, 2020-05-13:
I just noticed that for pads set to "from parent FP", FP to "from zone" and zone to "solid" or "thermal reliefs" seems to be working the way it should in 5.99, just for zone set to "PTH-only reliefs" I get a cute pattern of spokes to the thermal vias.

Regarding your question:
It would be (relatively) easy to switch the script generated footprints to solid, but we also have a significant amount of hand-made FPs with thermal vias, I think.
Plus, it's yet to be determined if doing so has unintended side effects and if the payoff is big enough to justify the hassle (if @evanshultz's wish is fulfilled, it would have to be done all over again for v6)

BryanKadzban, 2020-05-15:
5.99 works better?  I'm still on 5.1.5, but I'm OK waiting for 6.x if that's what it'll take.  And I do agree that it would be multiple levels of hassle if future releases also get the ability to put real vias into a footprint, or get more flexibility around the zone-to-pad connection options.

So let's call this bug unnecessary.  :)  Thanks!

closed 2020-05-15 by BryanKadzban