KiCad/kicad-footprints #2373: Footprint for SMD 5730 YUJI LEDS seems inverted

zackles, 2020-07-24:
Hello,
I just use the footprint LED_SMD:LED_YUJI_5730 and it seems it is inverted for his association with its symbol.

Datasheet is here :
https://cdn.shopify.com/s/files/1/0344/6401/files/YJWJ005-1.1_YJ-VTC-5730-G01.pdf?7088009637272241372

Could someone verify with me if it is indeed an error?

zackles, 2020-07-24:
[image]
Here is a picture making the comparaison between kicad output and datasheet

aris-kimi, 2020-07-24:
Hi @zackles , from the top view from datasheet image, polarity seems ok to me.

zackles, 2020-07-24:
Hi @aris-kimi, Thank you for your quick review. I agree with your interpretation. Will you update the footprint?

aris-kimi, 2020-07-24:
I will create a PR later if you do not want to give it a try. :-) But later tonight, so if you change your mind in the meantime, feel free to propose a fix.

zackles, 2020-07-24:
I may try to edit the footprint (not so hard I think) but I need to refresh my github skill ;p I'm looking at that right now.

zackles, 2020-07-24:
I don't understand why this previous PR did'nt corrected the problem
https://github.com/KiCad/kicad-footprints/pull/1948

aris-kimi, 2020-07-24:
Well... It did.. In my footprint libraries this footprint is correct. So i guess your library files ore not up to date..
This mean this issue can now close.. :-) Thank you for notice that..

zackles, 2020-07-24:
My Bad!  I have the 5.1.5 version. I'm installing 5.1.6 that include the corrected footprint

zackles, 2020-07-24:
Thank you aris-kimi for your help.
5.1.6 version works for YUJI_5730 LED.

closed 2020-07-24 by zackles