KiCad/kicad-footprints #2380: Layout guidelines for right-angle PCB edge connectors

calebreister, 2020-07-31:
I've been working on a script to generate Samtec right-angle Q Strip and Q Pairs connectors, and run into a questions regarding what the "correct" way to represent the PCB edge in a footprint:
1. Where should the courtyard end? Should it extend beyond the edge of the PCB?
2. What layer should the PCB edge be on? I've seen it on both `Edge.Cuts` and `Dwgs.User`.
3. If the edge is not in `Edge.Cuts`, should there be a label for the PCB edge? If yes, what layer should I use for the label? Should the label be placed on the inside or outside the edge?
4. How wide should the PCB edge line be in comparison to the connector? Should it extend all the way to the edge of the courtyard?
5. According to [F5.2 in the KLC](https://kicad-pcb.org/libraries/klc/F5.2/), the footprint name should be placed below the footprint. Does this rule still apply if the name would have to be placed outside of the PCB?
6. Where should the footprint origin be? Should it be aligned to the PCB edge?

Here's an example of one of the footprints I've generated using my script.
[image]

Related pull requests:
- #1304 
- pointhi/kicad-footprint-generator#397

evanshultz, 2020-08-12:
There are other librarians and many parts I'm not familiar with, but here's my suggestions:
1. Extending around the entire component body, including the part of the component off the board, is what I see in the library now.
2. `Edge.Cuts`. This is a limitation in the tool now, but I understand KiCad v6 will properly support cutouts in footprints. For now, I believe it will be supported by the script generator but you can't edit things later in the footprint editor.
3. `Dwgs.User` will work. Many USB connectors have text on this layer.
4. 0.1mm is what I see in existing footprints. I would say it should just be the width of the connector body. It will be up to the user to connect this to (or overlap it with) their board outline. That being said, I don't see any guidelines about this.
5. Yes. It will appear in the fab drawing which is fine.
6. The footprint origin should be where it always should be: pin 1 for THT parts and the body center for SMD parts (unless pick-n-place or another reason overrides this).

calebreister, 2020-08-19:
Thanks for the info. I'm currently reworking some of the code in my scripts.