KiCad/kicad-footprints #2381: SOT-143 footprint

matianfu, 2020-08-06:
I have check several documents for this package, including:

+ onsemi: https://www.onsemi.com/pub/Collateral/318A.PDF
+ diodes: https://www.diodes.com/assets/Package-Files/SOT143.pdf
+ nxp/nexperia: https://assets.nexperia.com/documents/package-information/SOT143B.pdf

Apparently, the sot-143 footprint in current KiCAD library is way too large. It is for wave soldering, not for reflow. Furthermore, Pad for pin 1 is misplaced. The edge is not well aligned. There is no reason to do so.

chschlue, 2020-08-06:
Are you willing to submit a fix?

matianfu, 2020-08-06:
I don't know if there is a standard? Including:

1. the line width of silkscreen, fab, and courtyard.
2. the fab and courtyard enlargement compared with the physical size and occupied area provided by the manufacturer.

Is there any agreement on those value? If so, I am happy to try it. :)

Also, the 3D model seems no appropriate for this package. All three documents listed in the first post agrees that the maximum distance between pin edges are no more than 2.5mm. But the SOT-143.wrl model seems to be at least 2.8mm width. I think this should be fixed either.

chschlue, 2020-08-06:
Short answer: you can use the line widths of the existing FP, they're fine; fab should be the nominal body outline while courtyard should be 0.25mm from fab and copper.
Long answer: https://kicad-pcb.org/libraries/klc/

If anything is unclear, feel free to ask.

perigoso, 2020-08-19:
I was working on this because i needed this package, i added a suffix to the originals `_WaveSolder` but it seems it's not done anywhere else, should we keep the old footprints? under what name? maybe specify the pad sizes

chschlue, 2020-08-19:
If there is demand (is SMT wave soldering still a thing?), I guess it would be fine to add `_WaveSolder` footprints.

perigoso, 2020-08-19:
I don't think it's relevant anymore, we don't have wave solder footprints for anything else, I'll submit the PR without them.