KiCad/kicad-footprints #2391: Through-hole fastener pad is on all layers instead of top/bottom (in multi-layer board)

StefanHamminga, 2020-08-14:
Through-hole fastener pads appear on all layers instead of top/bottom when using mount hole footprints in a multi-layer board. I'm not sure if this is by design or not, but it's wasting a fair amount of space. An 'offending' footprint is for example this one:

https://github.com/KiCad/kicad-footprints/blob/2742356730f9289f82ff5d895f67e9af5495141d/MountingHole.pretty/MountingHole_2.2mm_M2_DIN965_Pad.kicad_mod#L16

If it's agreed this should be top + bottom only I can probably make up some Linux shell-fu to bulk convert any line like above into two lines for the top and bottom copper layers. Would this be preferred or a pull request?

chschlue, 2020-08-14:
I don't think there can be a real solution until KiCad gains support for pad stacks.

IMO, pads on all layers is the best option we currently have. You may want to connect the pad to an inner ground layer, for example.
Besides, it's still a plated hole. How you close are you going to route inner tracks alongside it?

StefanHamminga, 2020-08-14:
I'd agree if it was just the plated hole and through hole pad, but for the "pad" type of mounting holes it's the whole screw head pad on every single layer.

I'd rather have my power plane beefed up locally than a random bit of ground plane on every layer.

My own library has another solution, using a regular size through hole annular ring and an extra pad on one side for the fastener head: https://github.com/StefanHamminga/kicad-footprints/blob/master/MountingHole.pretty/MOUNT_HOLE_M2.kicad_mod

That leaves a lot more space for routing and only places the large pad where you'd expect one to be. It would be great if the official libs have a similar (imho more elegant) solution so I can drop the custom holes.

chschlue, 2020-08-14:
Your second proposal seems reasonable to me. Still, I suggest adding them to the current ones instead of replacing.
Perhaps with a `_TopOnly` suffix or something.

closed 2020-08-16 by chschlue