KiCad/kicad-footprints #2414: Potentiometer_THT:Potentiometer_Runtron_RM-065_Vertical courtyard is not a closed shape

j9brown, 2020-08-25:
The DRC failed on parts using the "Potentiometer_THT:Potentiometer_Runtron_RM-065_Vertical" footprint with "courtyard is not a closed shape".

Upon closer inspection, it appears that the bottom left and bottom right corners of the courtyard for this part do not quite join the adjacent lines.  I have edited a local copy of the footprint to pull these corners into alignment and the DRC check passes as expected.

chschlue, 2020-08-25:
Thanks for catching this.
Are you going to submit your fix?

j9brown, 2020-08-25:
I suppose I could.  I should probably double check the dimensions to make
sure I shoved the boundary in the right direction (it’s a tiny change
though).