KiCad/kicad-footprints #2435: R_2818 footprint help

aris-kimi, 2020-08-30:
Hi everyone.

I need help in order to create footprint from [here](https://www.vishay.com/docs/30347/wshp2818.pdf).

ds:
[image]

By` SMD_chip_package_rlc-etc.py` script and editing `size_resistor.yaml `, pads created are not as datasheet recommends.

Also by editing `smd_chip_resistors_smd.csv` and running script, footprint created was ok but has main differences from other resistors.

How should i proceed?

cpresser, 2020-09-17:
I think there is no need for a `_Via??` variant. That is something the pcb-designer should take care of.

You already found the correct generator. The pad-sizes it outputs are different from manufacturer recommendations because it builds them according to the IPC spec.

Are there other 2818 shunts available from other vendors? If yes, we should start to compare them.
A quick search only resulted in this: https://www.seielect.com/catalog/sei-hcsm.pdf

Currently I think a generic `R_2818_7146Metric` is the way to go.

aris-kimi, 2020-09-17:
Thanks @cpresser , I need a few more days to finish with some issues. Weekend i think i will find the time.