KiCad/kicad-footprints #2449: Footprints for 2012 and 2008 capacitors

tpimh, 2020-09-04:
The closest that I found was 2010.

[image]

2012 is a wider version of 2010 and 2008 is a narrower version of 2010.

evanshultz, 2020-09-04:
Hello @tpimh . Is this a request for footprints? If so, our chip footprints are generated by a script at https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts/SMD_chip_package_rlc-etc.

cpresser, 2020-09-17:
@tpimh please add a link to that datasheet. 

Are there more manufacturers for those? Please do some research and post your results here. Then we can choose which one we want to include in the official lib.

tpimh, 2020-09-18:
This particular screenshot is from Vishay Vitramon safety MLCC series datasheet. Other manufacturers are not using these sizes, so probably a separate library should be created for them. While 2008 is still poular, it seems that [all parts in 2012 package are marked obsolete on digikey](https://www.digikey.com/products/en/capacitors/ceramic-capacitors/60?FV=-8%7C60%2C16%7C140983&quantity=0&ColumnSort=0&page=1&pageSize=500), but Vishay Vitramon come in various sizes:
[image]
I couldn't locate a datasheet that would describe them all, so probably I would need to download multiple datasheets and combine them together.

tpimh, 2020-09-18:
I did some research and looks like only 2008 and 2012 are exclusive to Vishay Vitramon (also 4044, but I could only find [one capacitor of this size](https://media.digikey.com/pdf/Data%20Sheets/Vishay%20Vitramon/VJ_CDC_Series_Rev_Jan_2018.pdf) and it's obsolete). I made this table which lists manufacturers that also use similar capacitor sizes:
Size | Rarity
-- | --
0201 | Common
0402 | Common
0505 | American Technical Ceramics, AVX, KEMET, Knowles, Venkel
0603 | Common
0805 | Common
1111 | American Technical Ceramics, AVX, Johanson Technology, Knowles, Murata
1206 | Common
1210 | Common
1805 | AVX, KEMET
1808 | AVX, Johanson Technology, KEMET, Knowles, Murata, Samsung, TDK, Venkel, Würth, Yageo
1812 | Common
1825 | Common
2008 | **Exclusive**
2012 | **Exclusive, obsolete**
2220 | Common
2225 | Common
2525 | American Technical Ceramics, Knowles
3040 | KEMET
3640 | Common
3838 | American Technical Ceramics, AVX
4044 | **Exclusive, obsolete**

cpresser, 2020-09-18:
So, only `2008` has a real use-case? And it is only available from vishay? 
Please add a link to the datasheet.

tpimh, 2020-09-21:
I'm not sure if it's a real use-case as it's only a few capacitors of this size only from vishay. I had no problems modifying 2010 footprint to make 2012, so 2008 shouldn't be a problem too. Here is the datasheet: [VJ Safety Certified Capacitors C0G (NP0)](http://www.farnell.com/datasheets/2699547.pdf).
I think I will later make a bigger table listing more capacitor sizes form different manufacturers with links to related datasheets.